prusa3d/Prusa-Firmware-Buddy #3192: [FEATURE REQUEST] Support for spool-joining on Prusa XL

ddavidebor, 2023-08-12:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - XL 2 to 5 extruders

**Is your feature request related to an existing problem? Please describe.**
When printing large prints on the Prusa XL, it's easy to run-out of filament. Please implement the possibility of having two or more spools of the same material, and automatically switching between between them when one runs out.

**Describe the expected functionality**
When a spool runs out in filament slot N, the standard prompt is triggered. if N+1 has the same filament type, a countdown will inform the user they have a limited time to switch to the new filament, otherwise it will be automatically switched over to N+1.

The printer can purge in between the build plate and the rear, then slide the extruder back and pick it up again to get rid of the piece of filament stuck to the extruder.

Preferably the behaviour is somewhat configurable through GCODE.

murk-sy, 2023-08-18:
There are a few ways to go about this:

1) Entirely relying on Prusa Connect and future material management features - would allow just loading materials and the printer figuring it out on its own 
   * +Easier to navigate interface
   * -Too cumbersome to be controlled efficiently on-machine
   *  -Requires cloud connectivity
2) Set an extruder as "Twin" on filament load (instead of material, select a tool that already has loaded filament, which marks the already loaded tool as Main and new one as Twin)
   * +Extremely simple to use
   * +Twin status can be automatically cleared on unload/runout
   * +Allows fairly straightforward addition of multiple twins (M+T or M+T+T, or M+T+M+T+T)
   * -If not used exclusively for same color it can cause miscolored prints
   * -Making another print with same gcode may require moving filament from one extruder to the other (imagine doing a 1,6kg print twice in a row)
3) During a print, set an extruder as "Twin" temporarily
   * +Useful when color doesn't matter but finishing the print in a timely manner does
   * +Temporary, prevents unintentional miscolored prints
   * -Cumbersome if you need it for every print

The more I think about it, implementing all 3 would likely be the best way to go about it. But just having 3 or 2 would already be a lot. To clarify, I've never used any similar system (or even MMU, if that supports it), this is just my impression of how it should work.

A bigger question is nozzle priming - it may require printing an otherwise unnecessary prime tower until prime into infill becomes a thing.

pacaj2am, 2023-12-22:
This is currently implemented. Just select multiple filaments for print.

robloh, 2023-12-22:
> This is currently implemented. Just select multiple filaments for print.

How would you know beforehand where to switch to the next filament in the gcode?

pacaj2am, 2023-12-22:
I think those are 2 separate questions - you can do that in GCode - but that would be on specific place - in the gcode privew right on the height range left or right click and add the change (https://help.prusa3d.com/article/color-change_1687#automatic-color-change). 

Spool join is bit different - it is doing the change automatically when filament runout detector triggers (you don't need to plan to specific point in gcode). It's done in filament mapping - you can select multiple tools for one filament:
[image]
and that will do the spool join. Beware - filament runout sensor needs to be enabled.

ddavidebor, 2023-12-22:
I'm closing my own issue as this is now implemented

closed 2023-12-22 by ddavidebor

burnpanck, 2024-03-24:
It would be great if spool-join could be enabled after-the-fact; specifically, when the runout sensor triggers, rather than at the start of the print. I had a case of a single colour print where print colour does not matter, and the filament did run-out unexpectedly. I did not have replacement filament ready, and would have liked to switch to another toolhead (or at least being able to unload the filament from another toolhead and re-load on the active one). But with the current implementation, I had to open a second spool of a color I already have on the printer, something I would have liked to avoid.

ddavidebor, 2024-03-31:
I think it’s best to open a separate issue @burnpanck