prusa3d/Prusa-Firmware-Buddy #3193: [BUG] <5.0 RC on Prusa MK4>

williamrehndell, 2023-08-13:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MK4 - Kit]

**Printer firmware version** - [5.0 RC]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Stock]

**USB drive or USB/Octoprint**
USB flash drive

**Describe the bug**
  A clear and concise description of what the bug is.

**How to reproduce**
 The 3D printer stopped during the print in the middle of the model, no error messages. The printer directed me to a screen when prints are finished. Self-tests of the printer were completed without any issues.

**Expected behavior**
The printer should have been able to complete the print without any issues.

**G-code**
Check attached files. 

**Crash dump file**
Check attached files. 

**Video**
Check attached files.

jindrichbenes, 2023-08-13:
It says files are deleted.
Anyway, are you attaching gcode from your desktop or the actual USB drive?
Please attach file from USB drive.

stsfin, 2023-08-14:
Are you using the prusa provided usb-stick? if so try with another one. those have alot of issues and most of these kinds of problems go away by using something better.

DRracer, 2023-08-14:
@williamrehndell Please provide the G-code file **copied from the USB flash** drive you used to print.

rtyr, 2023-08-16:
Thanks. We were able to reproduce it.

closed 2024-06-24 by github-actions[bot]