prusa3d/Prusa-Firmware-Buddy #3197: [ENHANCEMENT] Stop print after filament runout / unload

ulab, 2023-08-14:
**Printer type** - MK4

**Is your suggested improvement related to an existing problem? Please describe.**

I just had a filament runout and after the filament was unloaded, the printer waited for me to insert new filament. There was no option to stop the print instead.

I usually print some small items with "Complete individual objects" enabled to finish a spool. In that case I don't care for the rest of the objects to finish.

**Describe the expected functionality**

It would be nice if you could stop / cancel the print in that case without having to resort to a hardware reset.

larcin78, 2023-08-14:
This has happened to me several times. Super frustrating.

ulab, 2024-05-28:
I still think it would be a nice feature.

ulab, 2024-08-26:
It just happened again yesterday and I had to reset my XL to stop the print. This should not be necessary.

3d-gussner, 2025-11-05:
@ulab I assigned myself to prevent stale bot. Will try to get some feedback from the developers.

ulab, 2026-09-14:
Ah yes, the song of my people.

ulab, 2026-11-16:
Never gonna give you up...