prusa3d/Prusa-Firmware-Buddy #3198: new firmware 5.0.0-RC much slower LAN Upload speed than 5.0.0 Alpha 4

donnywhizbang, 2023-08-14:
The new firmware 5.0.0-RC made my LAN Upload via Prusalink much slower. With 5.0.0 Alpha 4 (Input Shaper) I had much faster LAN upload speeds.

TravisWilder, 2023-08-14:
I can confirm that RC is slower than A4

cristian-rivera, 2023-08-14:
I'm relieved to find out that I'm not the only one experiencing this. I can also confirm the same!

abjugard, 2023-08-15:
Also in agreement here.

Noticeable degradation in upload speeds that were already atrocious in 5.0.0-A4 and other previous firmwares.

marek26340, 2023-08-17:
Hello, could anyone here please clarify if this is related to both WiFi and Ethernet? Or only one of those connection methods? Asking before I upgrade from alpha4. Thanks!

abjugard, 2023-08-17:
I'm using Ethernet with mine. I'm pretty confident my first internet connection back in '97 was faster than this upload.

Area5142, 2023-08-18:
It is slow too when using LAN upload from PrusaSlicer - maybe because PS uses the interface from PrusaLink.

MarcelMalik, 2023-08-20:
Can confirm slow upload with ethernet.

Futrax, 2023-08-20:
Can confirm slow upload, Wifi.

jwkicklighter, 2023-08-21:
Using Ethernet with gigabit local networking, the upload speed is incredibly slow as compared to the last Alpha release.

drchslrk, 2023-08-21:
I've tried a walkaround with PrusaConnect. But sadly this doenst work either because the upload connection from PrusaSlicer to PrusaConnect doenst seem to be stable. So at least the only reliable option is to use the usb stick. 

@Prusa maybe we can get a CD or DVD slot in the MK5? :)

PrusaConnect Issue: https://github.com/prusa3d/PrusaSlicer/issues/11076

bobasaurus, 2023-08-21:
Same issue here.  I went back to alpha4 and the wifi speeds are good again.

abjugard, 2023-08-22:
> Same issue here. I went back to alpha4 and the wifi speeds are good again.

Could you measure and provide an estimate to how fast it is?

togrim99, 2023-08-22:
I did a tcpdump while doing an upload via prusa-link to the printer, and the average throughput was 40kbytes/s for release 5.0.0RC.
A quick glance at the packets shows that the MK4 is announcing a tcp-window of only 8192 bytes when the transfer starts, and this window is quickly filled up, and then struggles between tcp 0-window and 8192 (with lots of dup ACKs and retransmits). Didn't check into the code, but I guess the buffer size of the Ethernet driver is a tad to small, and the buddy board does not keep up freeing up that buffer.
Will revert back to a previous release and check if the tcp-window is the same and how the throughput is.

cristian-rivera, 2023-08-22:
I am currently not at my keyboard so I can't link to the specific change. However, when I initially encountered this issue, I analyzed the code differences between version 5.0.0 RC and Alpha 4. I noticed modifications in the TCP stack. I suspect these changes might be the root cause of these problems.

abjugard, 2023-08-23:
This is beyond ridiculous, I've been uploading a 15mb gcode for about 13 minutes now, and I'm only 63% done...

@Prusa-Support did anyone of you even try this firmware before uploading it?

MartinMajewski, 2023-08-25:
Soooo sloooooow. Over WiFi, I get around 100kb/s, but over Ethernet, it is a magnitude slower. Sending a 5MB file from Slicer to PrinterLink takes about 15 Minutes over wire.

For now, I connected an OctoPrint server to my MK4, but now I'm lacking any helpful information about the print status on the LCD screen.

mkatzNU, 2023-09-01:
I'm just adding another voice to the 5.00RC is considerably slow downloading files to the printer.  

Pre-5.00RC was slow 5.00RC is ridiculously slow.

My MK3 with a Raspberry PI Zero 2B was orders of magnitude faster than even the pre 5.00RC Firmware.  And now it's even worse.  Files that should take seconds to download are taking minutes.  And I'm using wired ethernet rather than WiFi to get the extra speed.

Isn't the MK4 supposed to be better than the MK3 is all aspects?

the-powl, 2023-09-02:
I just upgraded from 4.7.2 to 5.0.0-RC and uploads via PrusaLink over wifi got a whole lot slower... like.. 10 times slower.

mempf, 2023-09-02:
I wanted to post an accidental discovery I made in terms of WIFI file transfer speed on the MK4 with the 5.0 RC firmware. I have noticed that I get dramatically increased transfer speeds if I go into but don't select any option in the "Preheat" menu directly in the printer menu system. Does this occur for anyone else?

Futrax, 2023-09-02:
@mempf, same thing happens here! I have about the same speed as with 4.7.2

KBalage, 2023-09-04:
This is a good workaround for the upload speed, the only downside is that the printer report 'Busy' while staying in the preheat menu, so the print cannot be started.
I hope Prusa will solve this case soon, because it's pretty annoying overall.

togrim99, 2023-09-06:
I tried the workaround with the menu option "preheat" active and then doing the transfer, but it didn't make a difference, still about 40kbyte/s over Ethernet. Didn't try the WiFi yet.
But I digged a bit into the code, to see if there are any diffs between 5.0.0RC and previous versions. I discovered that the buddy board is using the lwIP - A Lightweight TCP/IP stack protocol suite, and the different options are set in include/buddy/lwipopts.h.
And there we have an interesting comment:

```
FIXME:
 * Workaround:
 *
 * We observed a very weird bug where, while sending a "file", the packets
 * either got reordered (1, 3, 2), which makes a mess of all the TCP
 * congestion algorithms or, when the other side ACKs twice the first
 * packet (to hint that the second one is missing while seeing the third)
 * before the second one goes out, we never actually send the second one.
 * Finding the real cause in either LwIP or our integration of that will be
 * *fun*.
 *
 * As a temporary workaround to deliver the damn file and not require F5 3
 * times to get the pages working, we make sure to send only a single
 * packet at a time. That way it won't reorder. This gives us slight
 * performance degradation, but that doesn't matter with the small web page
 * and makes it actually usable.
```
So I guess this won't be an easy task...

abjugard, 2023-09-07:
> This gives us slight performance degradation, but that doesn't matter with the small web page and makes it actually usable.

Hard disagree.

cristian-rivera, 2023-09-07:
@josefprusa mentioned in the podcast last week that 5.0 is going to ship this week with significant network improvements 🤞 

https://www.youtube.com/live/Yt3EIJ2_WuU?si=HLBF410lfJoBzhcM&t=10m55s

abjugard, 2023-09-18:
> @josefprusa mentioned in the podcast last week that 5.0 is going to ship this week with significant network improvements 🤞 
> 
> https://www.youtube.com/live/Yt3EIJ2_WuU?si=HLBF410lfJoBzhcM&t=10m55s

They have very long weeks in Czechia apparently.

TravisWilder, 2023-09-18:
> > @josefprusa mentioned in the podcast last week that 5.0 is going to ship this week with significant network improvements 🤞
> > https://www.youtube.com/live/Yt3EIJ2_WuU?si=HLBF410lfJoBzhcM&t=10m55s
> 
> They have very long weeks in Czechia apparently.

waiting and checking every day ...

TravisWilder, 2023-09-18:
lol .. just release .. for MK4

cristian-rivera, 2023-09-18:
No mention of fixed networking though 🤔

TravisWilder, 2023-09-18:
it seem faster to me
it will do on first  print a recalibration on the x/y (noisy)

TravisWilder, 2023-09-18:
first layer was perfect ... with final 5.0
I hope the 5.0 comes soon also for the XL

TeTeHacko, 2023-09-18:
Just uploaded 54MB file with 5.0.0 FW over ethernet, mikrotik says: Rate		100Mbps
Started Today at 2:53:43 PM Finished Today at 3:00:27 PM

Tx | cur: 18.5 kbps | avg: 641.9 kbps | max: 2.4 Mbps
Tx Packet | cur: 9 p/s | avg: 80 p/s | max: 287 p/s

its still very slow
kopiruje se to jak kdyz sere a maluje ...

arekm, 2023-09-18:
> Just tested version 5.0.0, and the WiFi speeds are significantly improved compared to the RC version. Uploading an 8.5 MB file now only takes 65 seconds (it previously took nearly 10 minutes).

That's still slow 134KB/s (and "10 minutes" was some new bug; earlier, before that bug, it was around 100KB/s over wifi). 

@TeTeHacko got very similar ~137KB/s over ... wired ethernet. That's even worse as people reported 300-400KB/s in April for wired ethernet: https://forum.prusa3d.com/forum/postid/650377/

marek26340, 2023-09-18:
@TeTeHacko C'mon, it's much better than how slow the RC was. If you want to complain about these speeds still not being good enough, either look up "sneakernet", or create a new issue. According to my calculations with the limits of the serial bus between the NIC and SoC in mind, your transfer should've taken right around 7 minutes, which it did.
@DominikPalo reported that he got up to right around 1Mb/s. Pre-RC speeds, nice!

If more people chime in and say that this specific issue from RC is fixed, this issue should be closed and new issues should be opened if needed.

abjugard, 2023-09-18:
We have a general issue #3049, this issue that deals specifically with the slowdowns caused by the RC should be closed, and further discussion moved over to that one.

Prusa-Support, 2024-01-19:
Thank you all and I must agree with the last comments.
This thread includes precious feedback but is rather specifically based on an old firmware pre-release.
Plus, we have other suitable existing issues for most of the reported problems.

This issue will be closed but remains linked to https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3049.

Michele Moramarco
Prusa Research

closed 2024-01-19 by Prusa-Support