prusa3d/Prusa-Firmware-Buddy #3200: [BUG] NIC submenus do not (contain the option to) load the static IP settings

TojikCZ, 2023-08-14:
**Printer type** - MK4

**Printer firmware version** - 4.7.2

**Original or Custom firmware** - Original

**Describe the bug**
Loading the Wi-Fi credentials does not load the wifi static ip settings
The option is missing entirely for Ethernet
I assume it made sense in code, but I had to go to the forum to figure this out.
The correct *load from the ini file* option that loads the static ip section is in Settings under System

**How to reproduce**
Try to set a static IP for Wi-Fi using prusa_printer_settings.ini and load the settings with the option in the Wi-Fi submenu, they won't load.
Set Ethernet to static - no option to load or edit the IP in its submenu

**Expected behavior**
  An option in the Wi-Fi submenu should load all settings regarding the Wi-Fi, not just the SSID and passphrase

**Video**
A video would show the loading of the settings and how zeroes end up loaded instead of the actual IP from the flash drive.

murk-sy, 2023-08-15:
Duplicate / related bug #2367

TojikCZ, 2023-08-16:
Yup, failed to find it 🙁 tried tho, but failed.

closed 2024-06-24 by github-actions[bot]