prusa3d/Prusa-Firmware-Buddy #3203: [FEATURE REQUEST] Add the Control tab to PrusaLink on the xBuddy platform

TojikCZ, 2023-08-15:
**Printer type** - [MINI, MK4, XL]

**Is your feature request related to an existing problem? Please describe.**
PrusaLink on the Pi has a Control tab allowing users to move the toolhead and pre-heat

**Describe the expected functionality**
Implement the control tab for the xBuddy platform

GRIFFCOMMca, 2023-08-21:
+1 for this feature, not sure why the PrusaLink in the embedded is so different..

closed 2024-06-24 by github-actions[bot]

jaakkorop, 2025-02-08:
Would be nice, after all Prusa does advertise that we can use the printer locally, but the local network interface just lacks a lot of features.

Often I'd like to start the preheat and then come to the printer for monitoring the start of the print. Especially for Core One I'd suspect the enclosure heating for difficult materials might take a while, so this feature becomes even more relevant for users not wanting the cloud connectivity.