prusa3d/Prusa-Firmware-Buddy #3205: [BUG] Loadcell test instructions are misleading

Failerson, 2023-08-16:
**Printer type** - MK4

**Printer firmware version** - 5.0.0 RC

**Original or Custom firmware** - Original

**Optional upgrades** - Prusa enclosure

**USB drive or USB/Octoprint** - USB/PrusaConnect

**Describe the bug**
During the load cell check, the printer sais "Tap nozzle on beep". The printer does not make the beep sound when sound mode is set to "silent".

**How to reproduce**
Set the sound mode in LCD menu > settings > user interface > sound mode to "silent", and make the loadcell test in LCD menu > control > diagnostics > test loadcell

**Expected behavior**
The printer should make the beep sound, even if the "silent mode" is used, or even better the text shown on the LCD during the test should be rewriten to "Tap nozzle after the countdown"

**G-code**
 none

**Crash dump file**
  none

**Video**

https://github.com/prusa3d/Prusa-Firmware-Buddy/assets/33588796/5091c93c-0249-4e17-8d79-3d0235f89a1e

marek26340, 2023-08-17:
I'd say this is expected behavior since you've set it to be silent, so it is silent. Also, it tells you when to tap the nozzle on the screen with a big "NOW" text.

Failerson, 2023-08-18:
It would be, if the instructions did not say it will make a sound, when it does not. That does not sound like a expected behaviour to me.

GRIFFCOMMca, 2023-08-21:
I agree, this is a bug, no pre-existing settings or memory of settings should be required when going through a wizard, so "remembering" the printer was set to silent is not a valid response to this bug, the text on screen should look at the sound settings and KNOW it cant beep which then leads to incorrect instructions (no beep will ever happen, the instructions are incorrect).

closed 2024-06-24 by github-actions[bot]