prusa3d/Prusa-Firmware-Buddy #3206: [BUG] HAEP MALLOC ERROR OUT OF MEMORY (MK4 5.0.0-RC)

papacrown, 2023-08-17:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK4

**Printer firmware version** - 5.0.0

**Original or Custom firmware** - ORIGINAL

**Optional upgrades** - None

**USB drive or USB/Octoprint**
  Prusalink and USB drive, initially thought the USB was causing the problem but have switched it out and error still shows up. 

**Describe the bug**
In the middle of prints, the mk4 goes to the red screen and I get the HAEP MALLOC ERROR OUT OF MEMORY. A restart gets the printer working again but most of my prints are failing. If I leave the MK4 on for a while, the error also shows up. I have factory reset the mk4, tried with WIFI off, Prusalink off, Prusaconnect off, and with different USB sticks/gcode files and still get the same error. It's been happening in all the Input Shaper Alphas for me. If I revert back to release firmware, error goes away. 

**How to reproduce**
Happens randomly. 

**Crash dump file**
[dump.bin.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12365319/dump.bin.zip)

**Video**
  Please attach a video. It usually helps to solve the problem.

tompi23, 2023-08-17:
Hello, I have exactly the same problem and the support says that they don't have an error reported on MK4, only on XL.
They want the printer for service.

papacrown, 2023-08-22:
So I contacted support, they said the following:

`This error is related to the printing over the network. The network memory gets filled and it causes this error. It should be fixed in the full release. For now, you can print over the flash drive and it should happen. You can even turn off the wifi to make sure. Please try turning disconnecting the printer from the network.`

tompi23, 2023-08-23:
Thanks for the useful information. No one wrote to me. I re-uploaded the firmware and the error didn't show up all weekend. Hopefully they will finally fix it and not say they don't know about the problem.

Prusa-Support, 2023-09-29:
Thanks for reporting.

Please notice that the network connection may only be one contribution to memory depletion, hence a cause among others.
A simple reset should free up enough space in the memory for the printer to continue to operate.

Allow me to close this issue as a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3131 and https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2351.

Michele Moramarco
Prusa Research

closed 2023-09-29 by Prusa-Support