prusa3d/Prusa-Firmware-Buddy #3207: [BUG] Filament unload unnecessarily heats bed

OJFord, 2023-08-17:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - [4.4.1]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Bondtech extruder]

**USB drive or USB/Octoprint**
  USB usually, PrusaLink when I can persuade it to work

**Describe the bug**
Turned on to unload filament, nothing else, was previously off & cold for days, nozzle _& bed_ proceeded to heat to printing temperature for the loaded filament.

**How to reproduce**
Have a filament loaded; Filament > Unload

**Expected behavior**
Nozzle to heat, but no need for bed to heat

**G-code**
N/A

**Crash dump file**
N/A

**Video**
I don't think it's necessary or would help?

murk-sy, 2023-08-18:
Related issues / partially duplicates: #706 #1331

OJFord, 2024-05-29:
I despise stale bots.

There is not going to be something to say every 60 days (unless someone starts working on it and has test builds to share perhaps).

That does not mean nobody cares any more or it's somehow invalid.

However, #706 is indeed a duplicate, and there's more there to help battle the bot (no comment there for 7 months either, but plenty of past comments to show that yes people do care).

closed 2024-05-29 by OJFord