prusa3d/Prusa-Firmware-Buddy #3209: [BUG] Printer starts a new print for the last printed item after power loss, even if it was in standby before

erikarenhill, 2023-08-18:
**Printer type** - Mk4

**Printer firmware version** - 5 alpha4

**Original or Custom firmware** - Original

**Optional upgrades** - None

**USB drive or USB/Octoprint**
USB / prusa link

**Describe the bug**
Printer was in standby, I updated firmware in a smart plug that powers the printer so the power was cut, when the smart plug turned on again the printer started a randomly named print job from scratch based on the last printed object. E.g. the printer thought it was resuming a job when the power was cut, but it was not printing anything by that time.

**How to reproduce**
Set the printer in standby, cut the power, and then bring it back. It will start a new print

**Expected behavior**
If the printer was not printing before the power cut, it should not print a new item, just get back to the same standby state as previous

**G-code**

**Crash dump file**

**Video**

CommanderRedYT, 2023-08-27:
I can confirm, this also happened to me after my smart plug (I have it for measuring power consumption) decided to reboot and turn off the power for a short period. I then came home to a half-done print.

murk-sy, 2023-08-29:
@CommanderRedYT Which version are you using? I believe the issue should be resolved with 5.0.0-RC, but it is not fixed in current stable 4.7.2

CommanderRedYT, 2023-08-29:
> @CommanderRedYT Which version are you using? I believe the issue should be resolved with 5.0.0-RC, but it is not fixed in current stable 4.7.2

I'm not sure if I updated it, it might still run stock firmware from factory (it's the kit version tho)

I can check when I'm home

hlipka, 2023-09-03:
I'm on 4.7.2. I can confirm that id even happens when you did not print anything before: just power on the printer with a USB drive inserted, and it will select the newest GCODE file there, showing you the 'do you want to print that?' screen. When you now turn of the printer and turn it on, it will start that print.
That's actually a big hazard, because it means that when your printer gets turned on for some reason (even if you do it manually to start a print next), it might just start one on its own, without manual acknowledgement.
I think this should be fixed in a 4.7.x firmware, not everyone will go to the first 5.0.0 version right away (I personally won't, for example)

LordRaptor, 2023-09-18:
I also had this happen today on the 4.7.2 firmware. There was also no filament loaded, but the printer didn't seem to care. This is a scary bug

erikarenhill, 2023-09-18:
I believe this is fixed in 5.0.0 final released today, at least according to the release notes.. but not sure

CommanderRedYT, 2023-09-19:
I hope so too. I am always turning off the power to my printer via the switch on my outlet because of this.

closed 2024-06-24 by github-actions[bot]