prusa3d/Prusa-Firmware-Buddy #3212: [BUG] MK4 Upload failed with an error http error 408 and Wifi disconnected

Elfe, 2023-08-20:
**Printer type** - MK4

**Printer firmware version** - 5.0.0-rc

**Describe the bug**
The printer was printing a large file from USB with input shaping and speed increased to 104%.
I Started uploading the next part of the project. An other file with a 9 h print time and about 28MB in size.
While the upload was running the PrusaSlicer 2.6 reported an error 408 and printer went offline. "PrusaLink: Error uploading file: , HTTP 408, body: `"
(No longer responding in the PrusaLink Session in Firefox nor responding to icmp ping requests)

**Expected behavior**
  Normal Upload and Wifi connection still working.

pljda, 2023-09-08:
same problem for me
**Printer** : MK4 w/ firmware 5.0.0-rc
Prusaslicer linked via PrusaConnect and PrusaLink to MK4 via network cable

Upload & print or Uploading the file lead to a regular crash mentionning "HTTP 408:  " (it happens usually at 99%).
looking at PrusaConnect, the printer is correctly connected.

Looking in the PrusaSlicer Printer settings, the connection test goes on & off.

moloroplayer, 2023-09-22:
I Have the Same Issue with Prusa Slicer sometime its Working and in Some Case its 25 Times Abording the Upload

Tnunnery, 2024-01-01:
Same issue here. Slicer 2.7.1, Mk4 firmware 5.10.

Icenor, 2024-01-01:
Same error. Restarting Router and/or Printer solved it.

2423m00ch, 2024-01-29:
Same here. Slicer 2.7.1, Mk4 firmware 5.1.3. All the tests say the connection is good but when Slicer tries to send the file it stops with the 408 error.

p3ck, 2024-03-25:
Ping - We now have a beta for 2.7.3 and not even an acknowledgement of the issue here.  Thanks

devjoda, 2024-04-22:
Why is this happening?

thomasloven, 2024-05-30:
I've been having this problem for a long time, but also random problems with a few other services (like being unable to log in to Autodesk Fusion).
Today I installed Malwarebytes and ran a scan. It found and quarantined something call "PUP.Optional.Spigot", and BAM! Uploads started working again immediately.

That fixed my problems both from Prusa Slicer (error 408) and via Prisa Link (timing out at 1-2%).

StayPirate, 2024-06-24:
It started happening to me as well. I've been using OrcaSlicer for months, but only a couple of days ago I started getting this problem. It happened after the upgrade from OrcaSlicer v2.0.0 to v2.1.0. That makes me thinking that maybe the problem relies in a common commit (or more) from PrusaSlicer and the diff of OrcaSlicer [v2.0.0..v2.1.0](https://github.com/SoftFever/OrcaSlicer/compare/v2.0.0...v2.1.0).
I'm on Archlinux and I installed OrcaSlicer from [AUR](https://aur.archlinux.org/packages/orca-slicer-bin).

StayPirate, 2024-06-24:
I confirm, I downgraded OrcaSlicer from v2.1.0 to v2.0.0 and the bug is gone.

There are 289 commits between v2.0.0 and v2.1.0... which might be too many to investigate which one got inherited from PrusaSlicer and potentially introduced the bug. But luckily OrcaSlicer upstream folks [published three more releases](https://github.com/SoftFever/OrcaSlicer/tags) between v2.0.0 and v2.1.0.

I'm gonna test each one of it to find where the bug start triggering, that will reduce the number of commits we should later investigate.

StayPirate, 2024-06-24:
I investigated the OracSlicer commits, and I found out that the culprit is https://github.com/SoftFever/OrcaSlicer/commit/cd6cd0786fc494d2f1e2fd2123e791f65fbcc97a. This is the commit which introduced the bug explained here in OrcaSlicer.

Unfortunately, I cannot find the funcion `void MainFrame::load_printer_url(wxString url, wxString apikey)` in the [PrusaSlicer src](https://github.com/prusa3d/PrusaSlicer), the code base probably diverged quite a lot.

In any case, I suggest the admins to move this issue in the [PrusaSlicer repo](https://github.com/prusa3d/PrusaSlicer).

danopernis, 2024-06-24:
From what I understand, apart from problems with malware and OrcaSlicer, this issue was last observed on 5.1.x series of firmware.

Can we please get a confirmation that this still happens on 6.0.2 ? And can we please make sure the firmware is actual culprit here? From what I understand, `HTTP 408 Request Timeout` can still happen on rare occasions, but if it happens consistently, we want to know about it. Ideally including the precise steps to reliably induce it.

closed 2024-09-02 by github-actions[bot]