prusa3d/Prusa-Firmware-Buddy #3213: [FEATURE REQUEST] PrusaLink embedded - Allow setting password for "maker" user

GRIFFCOMMca, 2023-08-21:
- Mk4 (and presumably the Mini and XL)
- PrusaLink (embedded)

- PrusaLink has a default user of "maker", the password is preset, unable to set to a password you may require.

- Need to be able to change the password for the user "maker" to anything the owner decides
- Side note, would be nice to be able to create new users (and passwords)

nocakeinc, 2023-11-12:
I'd like to support this.
I do not need a super secure password, the printer is contained in my LAN, barred from WAN access. When accessing, I don't always want to have to fire up my password tool. It reduces the usability of PrusaLink.

wseifert, 2024-06-10:
I also support this request

danopernis, 2024-06-10:
Although this is technically older issue than https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3559 let me close this one as a duplicate in order to not clutter the list of open issue :pray:

closed 2024-06-10 by danopernis