prusa3d/Prusa-Firmware-Buddy #3215: Y AXIS WILL NOT HOME

Padawg, 2023-08-22:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** -MK4

**Printer firmware version** - RC 5.0

**Original or Custom firmware** -ORIGINAL

**USB drive or USB/Octoprint**
USB DRIVE

**Describe the bug**
After installing RC5.0 the printer will not home the Y axis.  It just continually tries to home then I get the message on screen. X homes fine.  It did print one time after I first installed the RC5.0 firmware and I used the input shaper profile in PrusaSlicer.  I went through all the suggested fixes.. belt tension, bearing check, clear path on Y axis.  
The printer will home all axis with firmware 4.7 and acts fine.

**How to reproduce**
In my case, just install RC 5.0

**Expected behavior**
  A clear and concise description of what you expected to happen.

technogaijin, 2023-09-01:
Same problem here.  Y is the only one that will not complete properly, but it prints fine.   I fiddled with tightness of the belt, positioning, etc, but couldn't get it to pass that one test (all others are fine). The screen shows Y at 0 at various points and it prints fine, so I just skip the calibration tests, but it still nags me, which is irritating.

Prusa-Support, 2023-12-28:
> The screen shows Y at 0 at various points [...]

The printer only knows the real (0,0,0) coordinates after the auto-home. The auto-home is performed before every print so no worries if Y=0 in the "wrong" position after the printer rebooting, as the position will be adjusted at the beginning of the print.

.
We have not been able to reproduce the problem but a slightly loose belt, a wobbling belt holder, or even slightly slipping bearings will definitely cause inaccurate homing and lead to axis test failures in general.
As far as I can tell, so far most of the reported cases are hardware-related.
e.g. https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3090, https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3289, https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3295

On a side note, I seemed to notice a gradual increase of shifted Y bearings since the implementation of Input Shaper, Pressure Advance and Precise Stepping that enable fast printing.

Maybe it was an impression, or maybe sometimes, even if assembled in the right position, the Y linear bearings will simply not hold tight enough in place and will slightly move over fast movements. The front one may hit at the end of the axis too - at least, it will surely hit at the front end of the axis for the axis test.

Re-aligning the Y bearings and tightening them half/one a turn further should solve the problem.

Michele Moramarco
Prusa Research

closed 2024-01-24 by Prusa-Support