prusa3d/Prusa-Firmware-Buddy #3217: [FEATURE REQUEST] Set Printer as Ready (for Prusa Connect Queues) on the device itself after clearing the build plate

DIY-Machines, 2023-08-23:
Printer: Mk4 and probably the XL also.

It would be very convenient to be able to clear a completed print job of the build sheet and then be able to confirm on the printer itself that the machine is ready to accept and begin the next print job from the Prusa Connect print queue.

wizbongre, 2023-08-23:
Yep, I'd love that feature too - find myself getting mixed up between print jobs in PrusaConnect and print files on the machine itself.

chaos777b, 2023-08-25:
dito

kibu7, 2023-08-28:
That would be a nice feature

ammas09, 2023-08-29:
I agree, this seems both very useful and logical.  

When I'm at a specific machine it when I am most certain that it is cleared and ready to start printing a new job.  Needing to go to a web page and possibly select the wrong printer seems like a needles hassle and prone to mistakes.

drchslrk, 2023-09-21:
dito

BlackBird11, 2023-10-29:
this is possible unter control -> set printer ready
i would still recommend the printer gives me the option right after a print finished instead of "reprint, home etc." .. just add "printer ready" exactly there after a print is finished! thanks

closed 2024-06-24 by github-actions[bot]