prusa3d/Prusa-Firmware-Buddy #3219: [BUG] G-code slices ok but causes the printer to freeze and reset during printing

wizbongre, 2023-08-23:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK4 Kit

**Printer firmware version** - IS 5.0 RC

**Original or Custom firmware** - Original

**Optional upgrades** - N/A

**USB drive or USB/Octoprint** USB Flash drive (that was included with the printer)

**Describe the bug**
I self-assembled the printer and it has worked perfectly for over a month. In recent weeks it has started to demonstrate a number of issues that appear to be similar to those captured in bug #3156 , however I am seeing slightly different symptoms so wanted to raise a new issue.

For some prints, the printer will stop dead when in the middle of printing and perform a reboot, as if the reset button has been pressed. This happens every time I try to print affected G-code files, however it doesn't happen for all files.

In relation to issue #3156 I thought this was the cause (as I have also experienced symptoms as outlined on that bug) however the key difference for me is that the print fails when I save the G-code file direct to the USB drive on my Windows PC, so it does not appear to be an issue with OTA corruption.

I performed some testing to see if the file was being corrupted when using PrusaLink and PrusaConnect and can see the SHA1 hash is identical for the file no matter how I transmit it to the printer, either OTA or direct to USB, so it seems like a different issue to bug #3156 .

**How to reproduce**
 Try to print the attached G-code file and it should freeze after ~7minutes.

**Expected behavior**
The print should continue and complete as normal.

**G-code**
The affected G-code file is in the attached "Export G-code LFC.zip" file.
[Export G-code LFC.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12417371/Export.G-code.LFC.zip)

**Crash dump file**
The crash dump file is in the attached "dump.zip" file.
[dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12417377/dump.zip)

**Video**
 N/A

wizbongre, 2023-08-23:
I've rolled the firmware of my device back to 5.0 Alpha and one of the G-code files that was causing the reset has now successfully printed.

wizbongre, 2023-08-25:
Since going back to 5.0 Alpha, not a single print has failed.

ErwanAliasr1, 2023-08-29:
I have one gcode http://pubz.free.fr/crash.gcode that crashes the mk4 5.0RC in a few seconds, it seems to be related with the fact I enabled the "do not cross the perimeters" option cf https://forum.prusa3d.com/forum/input-shaping/gcode-produced-by-prusaslicer-2-6-0-crashes-the-mk4-5-0-0-rc/

closed 2024-06-24 by github-actions[bot]