prusa3d/Prusa-Firmware-Buddy #3220: [BUG] <Enter comprehensive title>

Ilddor, 2023-08-23:
**Printer type** - MK4

**Printer firmware version** - IS 5.0.0-RC

**Original or Custom firmware** - Original

**Optional upgrades** - N/A

**USB drive or USB/Octoprint**
  PrusaLink to USB drive but also USB drive alone

**Describe the bug**
 In the Info section of print statistics print time appears to be added double the amount of actual print time. I specifically have all gcode files I used so far saved and adding print times of those amounts to 16hours, however printer reports time spent printing as 32hours (1 day 8 hours). Also consistently prints of a bit over 1h (1h 7m) make it jump by either 2 hours or 3 hours in total time (3 is probably when the rounding just aligned the way it was rounding down in first measurement and up in second measurement.

**How to reproduce**
  Note print time in statistics, print something, observe double time added to overall amount

**Expected behavior**
  Amount added is accurate

**G-code**
  N/A

**Crash dump file**
  N/A

**Video**
  N/A

t-makaro, 2023-10-24:
I think this is a duplicate of #3151

Ilddor, 2024-06-26:
Tested it, it does look like it's fixed. Thanks!

closed 2024-06-26 by Ilddor