prusa3d/Prusa-Firmware-Buddy #3229: [FEATURE REQUEST] Spooljoin for XL Multi-Heads

ulab, 2023-08-28:
**Printer type** - XL

**Is your feature request related to an existing problem? Please describe.**
With the build volume of the XL comes a problem: Most manufacturers only offer 1kg spools that will not be enough for bigger prints.

For those cases the MMU has a "SpoolJoin" option to switch to the next available filament if one runs out. This should be a feature on the XL with multiple toolheads too. 

**Describe the expected functionality**
The XL should have an optional Spooljoin feature which selects the next tool with the same type of filament if filament runs out on the active one.

westranm, 2023-09-13:
Same as #3192 I believe. Totally agree and I heard this is already in the works from Prusa support, but I haven't seen the corresponding ticket indicating that.

ulab, 2023-09-13:
@westranm You are correct. I will close this, since the other already has more discussion on this topic anyway. I am not sure why it didn't show up in my search before.

Jo mentioned in a Prusa podcast/live episode that it's going to be in the next firmware along with some kind of "print setup" where you can assign different nozzles before a print start.

closed 2023-09-13 by ulab

pacaj2am, 2023-12-22:
This is currently implemented. Just select multiple filaments for print.