prusa3d/Prusa-Firmware-Buddy #3230: [BUG] After 1+hr into print toolhead one begins to cool between swaps while toolhead 2 does not.

hatesyourface, 2023-08-28:
**Printer type** - XL 2-toolhead

**Printer firmware version** - 4.7.2

**Original or Custom firmware** - Original

**USB drive or USB/Octoprint** - USB (PrusaSlicer 2.6.0)

**Describe the bug**
  After the first hour of printing on a multi-hour print toolhead one cools when docked while toolhead two does not. This results in each swap to toolhead 2 happening immediately as expected, but switching back to toolhead one means a delay to wait for the head to come up to temperature each time. This extends the end time on the print, as it will no longer meet the estimated time due to the unexpected wait cycles.

**How to reproduce**
Any multi-hour multi-tool print.

**Expected behavior**
I expect toolhead one to remain heated to the proper temperature while it is docked if it will be used on subsequent layers. 

**G-code**
  It's 10mb, not sure where to upload it to.

**Video**
If we really need one I can create one later. Thanks!

Sle77en, 2023-09-10:
yep i also have this problem

salmander, 2023-09-18:
I have this problem as well. I am not sure if it is a firmware issue or a PrusaSlicer issue. I have raised a bug in the PrusaSlicer repo here: https://github.com/prusa3d/PrusaSlicer/issues/11320

hatesyourface, 2023-09-18:
I noticed it on a multi color print this weekend before an hour had passed, but I dont know what which point precisely it had begun. Glad it's not only me.

closed 2024-06-24 by github-actions[bot]