prusa3d/Prusa-Firmware-Buddy #3239: [BUG] Filament Sensor detects INS again on high sensor values

the-powl, 2023-09-02:
**Printer type** - Prusa MK4

**Printer firmware version** - 5.0.0-RC

**Describe the bug**
  After a while of printing my filament sensor stops working properly – when unloading the filament it doesn't detect that there's no more filament present in the extruder any more. Recalibration doesn't help for long. I've noticed an issue with the interpretation of the sensor values. Normally we'd expect there to be a certain treshold value above which the sensor status switches from INS to NINS. But in my case, if the sensor value goes even higher, it switches back to INS again. So there's two thresholds apparently.

I tried to slooowly insert filament and observed the filament sensor values:
no filament present: INS / 1866562
filament slightly touching the sensor: NINS / 1043470
filament completely inside: INS / 28825 (could also go up to ~400000 when I wiggle the filament arount a bit tho)

**How to reproduce**
  Not sure if it can be reproduced with a foreign machine. After some prints I noticed that the filament sensor didn't work any more so I performed a recalibration. Usually no filament in the extruder results in sensor values of about 1860000. After some prints that value suddenly dropped to around 1180000 and the printer evaluated this as INS. So I performed a recalibration. After some more filament changes the sensor finally started to produce values of about 1860000 again when there's no filament present. Then I noticed the odd behavior.

**Expected behavior**
  There should be only one treshold value above which the sensor status switches from INS to NINS.

[image]
[image]
[image]

ericus256, 2023-09-03:
I encountered this as well, while troubleshooting a filament sensor that was also behaving as if it was "stuck" (it wasn't, but it needed calibration every few days). While watching the values I also noted erratic jumping between NINS/IS and not within the previously calibrated ranges for such.

ZATGamer, 2023-09-13:
I will add my +1 here. I am getting similar behavior with my filament sensor. On a side note, when I run jobs that have a color change at a layer, it becomes impossible to change the filament as the sensor won't trigger to load the new filament.

the-powl, 2023-10-06:
I can confirm that the Bug is still persistent in 5.0.0. And the original reason for the false readings of the filament sensor is that bits of filament get stuck in the sensor every now and then.

closed 2024-06-24 by github-actions[bot]