prusa3d/Prusa-Firmware-Buddy #3243: User-defined temperatures for filament changes

Harry-58, 2023-09-04:
**Printer type** -MK4   Firmware 5.0.0-RC

**Is your suggested improvement related to an existing problem? Please describe.**
I have a FlexFilament that is printed at 180-200°C.
When changing filament, the lowest selectable temperature is 215°C (PLA).

**Describe the expected functionality**
There should be a possibility to set custom temperatures for filament change or to select a filament with 180°C.

MichaelJLew, 2023-10-25:
I would also like to do this.
Is there a way for me to edit the pre-set filament change settings? That would suffice for me if it is too hard to allow me to change the temperature in an ad hoc way.

Heliosmaster, 2024-02-16:
Same here with my Mini+

KlausKraemer, 2024-02-16:
Look at issue #3657. I just added all but the same comment. Did you use Willowflex?

closed 2024-06-24 by github-actions[bot]