prusa3d/Prusa-Firmware-Buddy #3249: [BUG] 5.0.0 RC2 network (Wifi/Cable nextwork) download speeds is considerably slower (10%) than 5.0.0RC1.

mkatzNU, 2023-09-05:
**Printer type** - MK4

**Printer firmware version** - 5.0.0RC2

**Original or Custom firmware** - Original

**Optional upgrades** - None

**USB drive or USB/Octoprint**
WiFi/Cable Ethernet

**Describe the bug**
  Network download speeds are approx. 10% of 5.0.0.R1 on %.0.0 RC2

**How to reproduce**
   Install 5.0.0 RC1 Firmware
   Time the download of a large file
   Install 5.0.0 RC2 Firmware
   Time the download of a large file
   Note that the transfer time of 5.0.0 RC2 is approx. 10% of the transfer time of 5.0.0 RC1

**Expected behavior**
   Faster download speeds not slower

mkatzNU, 2023-09-05:
I'm sorry that is Alpha 4 vs RC.   Please forgive me.

TravisWilder, 2023-09-08:
to open same issue multi times does not help - double

closed 2024-06-25 by github-actions[bot]