prusa3d/Prusa-Firmware-Buddy #3252: [BUG] Error 414: URI Too Long on filenames longer than 76 letters

Linusten, 2023-09-05:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK4

**Printer firmware version** - [5.0.0-RC]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [None]

**USB drive or USB/Octoprint**
  print via USB flash drive

**Describe the bug**
  Uploading .gcode via the webinterface with names longer than 76 letters fails. Error message: "414: URI Too Long".

**How to reproduce**
  Upload a .gcode file with MORE than 76 letters as name.
  Working filename: 123456789012345678901234567890123456789012345678901234567890123456789123456.gcode
  NOT working filename: 1234567890123456789012345678901234567890123456789012345678901234567891234567.gcode

**Expected behavior**
  Upload successful.

gudnimg, 2023-09-06:
Just adding I created a similar issue in another repo: https://github.com/prusa3d/Prusa-Link-Web/issues/432

neuron-whisperer, 2023-09-12:
I've also experienced this issue.

The complicating factor is that the error isn't returned immediately - the web interface starts uploading the model to the printer and then stops at some arbitrary point: 15%, 30%, etc. And the message doesn't always appear immediately when the problem is discovered, so it just looks like the upload is blocked for some unstated reason.

At the very least, if the firmware truly cannot handle filenames over a certain length, then the length needs to be performed and an error needs to be reported before the upload starts.

JacobChrist, 2023-09-21:
There is related issue in the Slicer repo: https://github.com/prusa3d/PrusaSlicer/issues/10901

Area5142, 2023-11-16:
This error also applies to Prusa XL (4.7.2) and PrusaSlicer (4.6.1).

neuron-whisperer, 2023-11-26:
> The complicating factor is that the error isn't returned immediately - the web interface starts uploading the model to the printer and then stops at some arbitrary point: 15%, 30%, etc. And the [...]

+1 to this, and the solution. How difficult would it be to check the filename length at the very start of the upload and immediately report an error? The solution here is trivial and I wish that the Prusa development team would prioritize it.

neuron-whisperer, 2023-11-29:
This appears to have been (at least partially) fixed in firmware version 5.1. I just uploaded a file with this filename:

> GothicTowerMossPole-TheArgentRose-ModularB-Tighter_0.4n_0.2mm_PLA_MK4IS_6h16m.bgcode

...which is 85 characters, and the MK4 firmware accepted it and stored it, and is now printing it.

Limitations:

1) I don't know if this only applies to .bgcode files or if it also applies to .gcode files. I presume both since it would be weird to have different supported filename lengths for different file types.

2) I don't know if this also applies to filenames with spaces. I presume so.

3) I presume that filename lengths are still limited but that the limit has been increased to something more reasonable. In that case, I don't know what the new length limit is.

I might try some tests to evaluate these features - if I do, I'll report back.

Skaronator, 2024-05-29:
NOT STALE

JacobChrist, 2024-05-29:
I think the last time I uploaded a file with a long filename it corrupted the fat and I had to format my ad card. I grabbed the latest firmware at that time but I don't know if the issue is fixed.  I posted somewhere else about this as there are several open tickets related to this issue.

closed 2024-09-02 by github-actions[bot]