prusa3d/Prusa-Firmware-Buddy #3259: [BUG] Slow Uploadspeed via Prusaslicer to MK4 compared to 4.7.2

TheAlexcaf, 2023-09-10:
**Printer type** - Prusa MK4 FA

**Printer firmware version** - 5.0.0-RC

**Original or Custom firmware** - Original

**USB drive or USB/Octoprint**
 USB Flash Drive

**Describe the bug**
 Super slow upload speed via Prusaslicer to MK4 compared to 4.7.2. Around 34 kb/s. Same for upload via PrusaLink.

**How to reproduce**
Upload G-code via Prusaslicer/Prusalink. Observe upload speed.

**Expected behavior**
Faster uploadspeed i guess?

**G-code**

[V2 23,8-52,1-20,2_0.4n_0.2mm_PLA_MK4IS_2h14m.txt](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12568027/V2.23.8-52.1-20.2_0.4n_0.2mm_PLA_MK4IS_2h14m.txt)

BigBen2003, 2023-09-11:
==English==
Transferring a 4.9 MB GCode file via Prusa Link takes all of 7 minutes.
==German==
Die Übertragung einer 4,9 MB großen GCode-Datei dauert über Prusa Link ganze 7 Minuten.

TheAlexcaf, 2023-09-11:
@MrJones4u But why was it so much faster on 4.7.2? I guess if the USB was the issue, then there should be no difference to 5.0.0 RC, right? On 4.7.2 litteraly every file uploaded in like 20 seconds max. On 5.0.0 it's minimum 1 minute for small objects, and for larger files, as you said, upwards of an hour.

MrJones4u, 2023-09-11:
Same reason; they are working on the firmware and e.g. reducing the memory available for buffering will limit the initial burst.

abjugard, 2023-09-15:
It wasn't acceptably fast on 4.7.2 either. This has been reported in other issues here, and Josef said on a stream that there was firmware with significant network improvements being released last week. Obviously that didn't happen but hopefully the release isn't too far away if it was supposed to be released a week ago originally.

marek26340, 2023-09-19:
The 5.0.0 final stable has been released. Please update to that and try again.

According to my testing, this has been fixed and is back up to pre-RC speeds (1 Mb/s).

TheAlexcaf, 2023-09-22:
> The 5.0.0 final stable has been released. Please update to that and try again.
> 
> According to my testing, this has been fixed and is back up to pre-RC speeds (1 Mb/s).

The Issue still seems to persist. I flashed to 5.0.0. Using Prusaslicer 2.6.1 i now get uploadspeeds of around 300kb/s, so its faster than RC but still way slower than your 1MB/s.

arekm, 2023-09-22:
His is 1Mbit/s, you want 1Mbyte/s.

abjugard, 2023-09-22:
> His is 1Mbit/s, you want 1Mbyte/s.

It's 2023, we should have at least 100mbit regardless of whether it's wired or wireless.

marek26340, 2023-09-23:
> > His is 1Mbit/s, you want 1Mbyte/s.
> 
> It's 2023, we should have at least 100mbit regardless of whether it's wired or wireless.

Then just use OctoPrint. Atleast for now. I think there's gonna be a hardware bottleneck at this point.

danopernis, 2024-05-31:
Closing this because the upload speed regression that was originally reported was fixed. While debating hardware choices is certainly entertaining, there is not much to be done in the firmware itself and I would rather spend my time more productive.

closed 2024-05-31 by danopernis