prusa3d/Prusa-Firmware-Buddy #3260: [BUG] Print Fan error at 2nd or so layer but all testes passed

davcin, 2023-09-10:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - 4.4.1

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint**
  USB flash drive

**Describe the bug**
  After updating the Firmware to 4.4.1 I started to get the Print Fan not on error. It wasn't happening before the update.
  Done the self test and the fan testes and the fan works (all tests passed).
  This error stops printing since it always fails.

**How to reproduce**
  Besides updating the firmware and try yo print I am not sure. I am sure it doesn't happen to everyone so I assume it must be a specific case.

**Expected behavior**
  Print without error.
  For this, I disable the fan check, not sure if it is a good idea but no alternative.

**G-code**
  Any G-code I tried.

**Crash dump file**
  (none)

**Video**
  (none)

laird2205, 2023-10-22:
I have the same issue. Started after going to 4.4.1+4194. If I am looking at it correctly, the previous firmware on my usb stick is Buddy_4.3.4+2835.bbf .

m-klecka, 2023-10-23:
Same issue on 5.1.0 alpha2, but was happening before too. Self tests show everything ok.

evxjson, 2023-11-16:
One of our printers with 5.1.0 alpha2 has this issue. All self tests shows ok.

Moyrn, 2023-11-24:
same issue here. i was on 4.4.0 all fine since months.
moved to 4.4.1 to instantly jump to 5.1.0 (eager to test the input shaper)
> impossible to print : print fan error

Self test is also KO on the print fan on my side as well

Moyrn, 2023-11-24:
Strange thing : I removed the print fan and the self-test was still KO - though RPM still reportingI then noticed the yellow wire was quasi loose. touching it and it was brokeni then resoldered it, and made another self test in free air : passed OK - reporting something like 7800RPMi mounted it back fully : passed OKI renabled "check fan"I launched a print : passed OK

So possibly a pinched or loose cable in my case. Firmware is just performing tests which were not there before.

tjjos, 2023-12-05:
Same thing is happening to me after I upgraded to 5.1.0. I really don't want to disable the fan check. Is there any other solution for this?

Moyrn, 2023-12-05:
on my side i fixed it by resoldering a faulty yellow wire on the fan. so not a firmware issue, but a less permissive firmware - i guess

tjjos, 2023-12-05:
That's definitely not my issue, as the fan is spinning while printing, but the error keep popping up.

Moyrn, 2023-12-05:
ok ok.

On my side that was also what i had.
fans were running fine, even in my hand. I could see the RPM very high
during the fan test...
It was spinning then indicated "failed". i was puzzled.

So i checked, and saw that the yellow wire was "almost" loose. When
removing the sticker, the yellow wire got out.

So i suppose the fans are ok - but the RPM read value (yellow wire) raises
issues.

Sure sure. maybe not your case. there are several cases bringing to that
issue.
but i wanted to be clearer

Makeneer, 2024-06-09:
Same issue on MK3.5 after upgrade from MK3S+

I have upgraded my Prusa i3 MK3S+ with MMU2S to Prusa i3 MK3.5 with MMU3.
Before the upgrade I had no problems with my printer and I have printed all necessary parts for the upgrade with no problem at all.

After the upgrade everything looked fine, the self-test passed with everything OK and printing was no problem until a couple of layers where printed.
After a couple of layers my printer gave an error on the print fan. I have checked the wiring and connectors, checked the working of the fan with the available menu options and did the fan test again.
Everything seems to be OK.

But my printer is still giving the same print fan error after a couple of layers.
I have changed the slicer profile so the print fan is running from the first layer and made a new test print.
The fan is running OK from the start of the print but after a couple of layers the printer is giving me the same error again.

I have checked the working of the print fan over the complete X-axis range a couple of times and it is working OK every time and every test.
e

closed 2024-09-02 by github-actions[bot]