prusa3d/Prusa-Firmware-Buddy #3261: [ENHANCEMENT] Missing option for custom NTP or NTP via DHCP on Mini

KaiserSoft, 2023-09-10:
**Printer type** - [MINI]

**Is your suggested improvement related to an existing problem? Please describe.**
This creates a problem because files already on the SD card have the correct time but anything uploaded over PrusaLink has a date of 1970. This problem makes Prusa Link unusable if you don't allow your devices Internet access.

**Describe the expected functionality**
Current firmware for the Mini requires Internet access or the printer will not have the correct time.
The firmware does not allow the configuration of a custom (local) NTP server FQDN/IP and it does not support using the NTP provided by DHCP either.
Both options should be supported.

This is a know issue but no one bothered to fix it. https://help.prusa3d.com/article/ntp-server-mini_151132
Hopefully this request will put it on the table.

xtomasnemec, 2023-09-12:
And maybe also add the non-standard time zones like +5:30 or +5:45

Prusa-Support, 2023-10-07:
Thanks for your suggestions.
However, these partially duplicate existing issues https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/639, https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1779, https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2317, https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3141.

I suggest closing this one issue and sharing your relevant piece of feedback on the related existing issues.
Scattered feedback over several issues doesn't quite help us track the popularity of the community requests and define development goals accordingly.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-10-31:
It seems safe to close this issue as a duplicate of mixed already posted issues.

Michele Moramarco
Prusa Research

Seaborz, 2024-07-04:
How is this solved in the linked issues? There is still no option to use your own NTP server via a setting or DHCP option. In the meantime this looks like a phone-home functionality of the printer since it uses a specific NTP server 195.113.144.238 which resolves to tak.cesnet.cz

Contents of /lib/WUI/sntp/sntp_client.c
`
/* TODO: enable DNS for ntp.pool.org as default sntp server*/

    // TMP: ip of Czech CESNET NTP server tak.cesnet.cz
    if (ipaddr_aton("195.113.144.238", &ntp_server)) {
        sntp_setserver(0, &ntp_server);
    }
`

Prusa-Support, 2024-10-12:
This is a quite old issue but thanks for the feedback. I guess the issue should stay open for now.

Michele Moramarco
Prusa Research

CZDanol, 2024-10-14:
Internal ticket: BFW-6184

CZDanol, 2025-04-17:
Since 6.2, we are now at least getting the NTP server through DNS (prusa3d.pool.ntp.org). Not a solution to this ticket, but now you can at least set up a DNS redirect.

hman2, 2025-04-17:
"but now you can at least set up a DNS redirect."

Only if you are an admin... Ordinary employees cannot set up DNS redirects. And you'll have a hell of explaining to do for your security department why you want to tamper with their DNS...

grimbeaver, 2025-05-05:
It looks like setting LWIP_DHCP_GET_NTP_SRV to 1 would enable getting the NTP server from DHCP.  It's a compile time option though, not something that can be turned on/off on demand.  But all the support seems to be there.

PTrUNB, 2025-09-02:
Rename the title to include Core One ?

zeus86, 2025-09-07:
i seriously don't know what the problem is, to   
a) just make this configurable in the first place and
b) respect NTP-Options that were delivered by DHCP  
It makes absolutely no sense to rely on some hardcoded pool in the internet. If NTP is delivered as DHCP-Option, then it is OBVIOUSLY the right choice for any given network, because that is what the network-admin intended it to be. when this fails, it is safe to assume, that your gateway may also provide NTP functionality. This is true for ~90% of all home-networks. if that also fails, AND you have access to internet, you can still fall back to the only reasonable ntp-pool on the internet: pool.ntp.org. Making this hardcoded was an error in the first place and definately should be fixed.

LasseR15, 2025-12-29:
For anyone interested in a workaround until it's implemented in the firmware:
You can overwrite the dns entry for `prusa3d.pool.ntp.org` in the dns server that your printer is using (i.e. point an A record to your desired IP address). After that, the ntp server you specified is used instead of the default one.

zeus86, 2025-12-30:
> For anyone interested in a workaround until it's implemented in the firmware: You can overwrite the dns entry for `prusa3d.pool.ntp.org` in the dns server that your printer is using (i.e. point an A [...]

Thats not a workaround, thats plainly the wrong way of doing it. 
While this may work for some setups, it certainly does not in all. While you generally don't want to mess around with DNS for Internet-reachable domains (except for fighting spam like with Pi-Hole), this comes with some caveats:

* not in every environment you can or are allowed to mess with the DNS setup
* this will break if DNSSEC or DoH is involved (and will not work properly in setups where the DNS always tries to fetch authorative answers)
* Prusas NTP aside for one moment: Most other thingys that communicate over the network use HTTP(s) of some sort for most of the things. When dealing with SSL (as e.g. in http"s") dns-manipulation may have very unintended side-effects.
* your local dns-server may not act properly as a DNS-Pool as expected
* you don't want to add  wierd exceptions for every single devices that does not adhere to basic IP-Networking-standard-rules, expecially because stuff like this can break other things that are otherwise unrelated (this becomes much more pronounced when dealing with RFC1918-IPs instead of hostnames)

The solution for this is easy and is like I previously stated: If an administrator cares enough to give you an NTP-Server via DHCP, do the right thing and use it as your NTP. You won't just choose your DNS-Server, gateway or IP-address randomly because "you're feelin' lucky". This is what the network kindly gives you so use it or don't participate in the network. That's how it is meant to be! What really annoyes me on this issue is, that it is an ultra-simple fix, that could be implemented yesterday if someone would took it seriously enough to just fix it the right way.

PTrUNB, 2025-12-30:
This is in my opinion a BUG not an enhancement since all the time and dates in the Printer are fubar if the NTP is not available / restricted. And No options are available to set the time and date or NTP server manually.

hman2, 2026-01-09:
"Rename the title to include Core One ?"

We have a MK4S/MMU3 and it has the same problem...