prusa3d/Prusa-Firmware-Buddy #3264: [BUG] Insufficient cleanup procedure when using PETG, bad first layer

szymonpk, 2023-09-13:
**Printer type** - [XL]

**Printer firmware version** - [4.7.2]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [none]

**USB drive or USB/Octoprint**
  USB

**Describe the bug**
  When I am printing with PETG, the nozzle cleanup procedure is insufficient to remove filament residue. It sticks to the nozzle and affects the first layer. I have to clean up the nozzle right before 'absorbing heat' ends. The picture is attached at the end.

**How to reproduce**
  Print with PETG on XL.

**Expected behavior**
  Perfect first layer every time, without manual intervention.

**G-code**
  https://gist.github.com/szymonpk/d0c715c53b2bb1a8bc5eaae714f8e9ed

**Crash dump file**
  n/a

**Video**
  [image]

bobasaurus, 2023-09-17:
This sometimes happens on the MK4 with PETG as well.

mperino, 2023-10-27:
The Nozzle Cleaning procedure on Mk3.9 and MK4 also leaves room for improvement with PETg.
On MK3.9 I have done extensive testing of several solutions where PETg fails to clean on a worn (some tearout on the purge line):
https://forum.prusa3d.com/forum/original-prusa-i3-mk3-5-mk3-9-assembly-and-first-prints-troubleshooting/petg-fails-nozzle-cleaning-every-time/

There is a similar thread in the MK4:
https://forum.prusa3d.com/forum/english-forum-original-prusa-i3-mk4-hardware-firmware-and-software-help/nozzle-cleaning-failed-error/

I believe that there are several interconnected issues here around the cleaning process when using PETg.
Im running the  2.7.0 Alpha 1, with the latest 5.1.0-Alpha2 firmware which has significantly improved it for me.  However it is still not reliable enough even with trying lowering temperatures. 

I've also tried several other fixes that are documented in the threads above.

One consideration I'd like to mention for any further fix is to account for older textured sheets. Especially ones where the rest of the sheet is in good condition, but the purge line has some tear out. I'd really love it if we could define Clean & Probe for (Left, Middle, Right) as a menu option.

szymonpk, 2024-05-30:
It is still an issue, although not a big one with 0.4mm nozzle, but my printer was shipped with 0.6mm.

ltlowe, 2024-07-16:
I am also having problems with correct 1st layer height for PETG with a 0.6 nozzle on my mk4. FW 6.02.   I have to manually clean the nozzle with pliers just before bed probing if I can time it right, otherwise the first layer is to high and I have a lot of adhesion issues on textured, satin, and PA sheets.  I haven't tried PETG on my smooth one.

I have had my printer for a couple months and so far have not had this problem with PLA.   I don't print much with nylon, but it seems to be ok when the sheet is clean.  I haven't used any other high temp filaments yet. 

As for nozzle size, I have been doing a lot of 0.6 printing lately, and can't say if it happens with smaller sizes.

koprivajakub, 2024-08-06:
Same here, if I clean the nozzle manually it goes well and I get a perfect 1st layer. But it is annoying as I cannot print on MK4 remotely, which is one of the reasons I upgraded from MK3S+

TBH: not sure how could this be resolved as I do believe that the printer needs to be heated to measure the distance correctly (due to the thermal expansion of the metal parts). But when heated small amount of filament will ooze from the nozzle and since the temp is not too high (170°C for the probing) it won't purge itself, but only squish as a very tiny blob on the nozzle and cause the probing to cause the error.

closed 2024-10-15 by github-actions[bot]