prusa3d/Prusa-Firmware-Buddy #3267: [BUG] Probably incorrect temperatures - Print stringing

xcybplx, 2023-09-15:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK4

**Printer firmware version** - Latest stable + lastest RC

**Original or Custom firmware** - [Original]

It has been reported by many users, that exactly the same print, using the same filament, in the same room (so external conditions are the same) prints perfectly fine on mk3s but not on mk4. At the moment lowering temperatures helps with the problem but here is the question:
Are we sure the temperature on the LCD screen is exactly the same as on the nozzle? Is it reported correctly? Maybe there should be or there is wrong offset?

Also, the bed temperature looks better on mk3s than on mk4. Even if the measure method is not professional, it still raise a question, why there is a difference.

Please take a look here for more details : https://forum.prusa3d.com/forum/english-forum-original-prusa-i3-mk4-how-do-i-print-this-printing-help/mk4-pla-stringing/

I hope someone from Prusa team can at least check it and contact those people with some solution.

tech-geek-4, 2023-09-18:
I do have exactly the same problem.

I tried the following: 
- useing a brandnew spool of prusament delivered with the new printer
- lowering the temp which doesn't do anything until it clogs
- different settings of retraction length and speed - no changes
- raising the temp - strings seem to get thicker
- wipe on retraction, no change
- z hop on/off - no change

PatrykSikora1, 2023-09-18:
I do have exactly the same problem on MINI+
I created thread on reddit where I am setting 230 degrees on two MINI printers and I am measuring real value with thermocouple that I put instead of filament at the very ent of the nozzle.
[https://www.reddit.com/r/prusa3d/comments/16lprai/extruder_set_temperature_not_matching_measured/](url)

In my case, when I set temperature to 230 degrees, thermocouple shows around 245-250 which is way off

feutl, 2023-09-24:
Based on the measurement I did today my bed temp is also much lower than the own showing in the display and the interesting thing is, that I do have bed adhesion problems since firmware 5.

Before that, everything worked fine and there was no need to measure the bed temperature with an external probe, but since firmware 5 and the new prusa slicer profiles I constantly have warping with the standard settings on PLA.

Setting the bed temp to 70degrees helps, but I did not had to do that with <5 firmware or the old prusa slicer profiles. It still needs some more testing, but it is somehow strange this appeared with out of a sudden ;)
Filament is the same, printer is the same as some weeks ago and the bed is clean, that is for sure :D

devbyrd, 2023-09-24:
I'm having the "wispy stringing" issues since building my MK4. Love the printer, but this is a REAL problem. It's affecting a ton of users (check the prusa forums). Is there any acknowledgement from Prusa developers that something is wrong with profiles or firmware?

xcybplx, 2023-09-24:
> Based on the measurement I did today my bed temp is also much lower than the own showing in the display and the interesting thing is, that I do have bed adhesion problems since firmware 5.
>
> Before [...]

Interesting, I never had firmware below 5. I started with rc5 if I remember correctly. So we don't know if it's firmware or firmware + slicer...

jhenley01, 2023-11-10:
I'm also seeing a similar issue. I just assembled my second MK4 and when I attempted to run a gcode file that prints great on my first machine I can see that it's printing much hotter. The part printed on the new machine had much more shine to it, I dropped the nozzle temp 15 degrees and it printed as expected.

I was then able to confirm the 15 degree difference with a thermometer. 

This is unacceptable for machines that are supposed to be reliable farm printers. I should be able to run the same gcode without having to have separate profiles for each one.

PatrickSannes, 2023-11-15:
I currently experience the same problem. Brand new filament (prusament) and a furry print. Never had any problems om my mk2.

If needed I'm willing to help testing/diagnosing.

bgiot, 2023-11-23:
firware 5.1.0 with slicer 2.7.0 doesn't seem to solve the problem.... probably a hardware issue with thermistor?

xcybplx, 2023-11-24:
After some deep investigations from user side, it turned out there are 2 versions of hardware. Can it be fixed by software? I don't know and Prusa is not responding. Anyway - here is the explanation why some machines are "faulty":

https://forum.prusa3d.com/forum/postid/685235/

In the next posts there are also pictures, measurements etc.

andreacampanella, 2023-11-26:
For me the fix was to lower every tempterature by 20 degrees, I think they must have run into a supply chain issue and the replacement thermistor is off by about 20 degrees

ra-kal, 2023-12-21:
> Based on the measurement I did today my bed temp is also much lower than the own showing in the display and the interesting thing is, that I do have bed adhesion problems since firmware 5.
>
> Before [...]

Last weeks I'm also experiencing PLA and PETG to warp badly, when I printed larger objects. I noticed the bed temperature may be lower than set. I was discussing this problem with technical support and they gave me replacement heated bed. I replaced the bed and I measured the temperature using external thermometer and it read 55C instead of 60C - the sensor was placed between the heatbed and the steel sheet. First print and there was slight warping of one edge again. The sheet was clean with dish soap. Next time I will clean it with IPA to make sure the sheet is perfect clean.

I'm not sure when it started, but now I'm on the latest FW 5.1.2.

pauserratgutierrez, 2023-12-21:
Wtf is happening with Prusa, wasn't expecting all this troubles.

andreacampanella, 2023-12-29:
new hardware, new software, growing pains

closed 2024-06-25 by github-actions[bot]

Ondrashek, 2025-06-27:
My Core One, converted from the MK4, does exactly the same. I press NO, and it always loads the correct color based on what I set in PrusaSlicer.