prusa3d/Prusa-Firmware-Buddy #3268: [BUG] Server error when uploading a file with existing file name.

nachtbraker, 2023-09-15:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [XL]

**Printer firmware version** - [4.7.2]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [-]

**USB drive 

**Describe the bug**
When uploading a file from Prusaslicer (2.6.1 in my case) to the XL, and that filename already exists, I get an error:

PrusaSlicer has encountered an error
Error uploading to print host:
HTTP 500: 500: Infernal Server Error
Unknown error
[image]

**How to reproduce**
Upload a file and print it.
Upload it again.

**Expected behavior**
Overwrite the existing file, perhaps with a warning.

**G-code**
N/A

**Crash dump file**
N/A

**Video**
N/A

nachtbraker, 2023-09-15:
Also note the spelling error "Infernal" instead of "Internal", but I believe this has been tackled in other issues.

Prusa-Support, 2023-09-16:
Exactly. Thanks for your suggestion though.

This issue will be closed as it duplicates https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2447.

If you are experiencing this frequently because you frequently work on project reiterations, consider adding a _variable_  like `{timestamp}` to the _output file name format_ in Prusa Slicer so that filenames will always be unique (mind to dump old files every once in a while).

Michele Moramarco
Prusa Research

closed 2023-09-16 by Prusa-Support