prusa3d/Prusa-Firmware-Buddy #3269: [BUG] Z-axis moves in small chunks when moved manually

the-powl, 2023-09-18:
**Printer type** - MK4

**Printer firmware version** - 5.0.0

**Describe the bug**
  When I longpress the rotary knob to get in the menu for manual Z axis movement and move the Z axis by a larger amount (let's say 50..100mm) it starts moving continuously but then after a few mm it introduces some form of stuttering. it's like the pathplaner doesn't consolidate the movements any more. The axis moves by 1..2mm then stops, then moves again by 1..2mm then stops.. and so on.

**How to reproduce**
  Lonpress the rotary knob to get into menu for manual Z axis movement. Turn the rotary knob by quite an mount to tell the Z axis to move up or down by 50..100mm.

**Expected behavior**
  Z axis should move mostly smoothly without perpetual stops.

marek26340, 2023-09-19:
Let me add to this by saying that the first ever autohome that I did after upgrading to the stable 5.0.0 release was definitely acting a bit weird.
After waiting for it to finally autohome itself the first time, it did not do that the second time. Did a selftest just to be sure and all the issues seem to be gone.
So no, I am not seeing your issue on my printer.

the-powl, 2023-09-19:
The first autohome where it did the x and y calibration was a bit noisy (as advertised) but when I now perform an autohome it's just fine. The z-axis problem issue is still persistent, also after a restart. Interesting that you can't reproduce it.

marek26340, 2023-09-19:
I'll go and try again, but this time I'll do it by holding down the knob. Will edit this comment with results in a couple of minutes™.

murk-sy, 2023-09-20:
I've had this kind of behaviour happen on the Mini on 4.4.1 stable, but I was not able to reproduce it reliably.

I believe it happened when running first layer calibration twice in a row with some specific order of actions, but I honestly can't say for sure.

Tupson444, 2023-11-14:
I have the same issue - when manually moving Z, it constantly moves bit by bit with stops in between (as if instructing it to move 1mm at a time).

This happens when moving by holding the knob in the main menu, but also when moving from the move axis menu, only a bit differently: it moves smoothly 10-20mm, then stutters for a couple of mm, then continues moving smoothly.

marek26340, 2023-11-14:
According to the patch notes for firmware 5.1.0-alpha2:
> Manual z-axis movement (using the long press shortcut) is now smooth and without stuttering. (MK4)

Been daily driving this version and I think it works perfectly fine for me. Although I'll admit that I haven't tested the long press Z adjust function on this version, since I practically never use it. Feel free to try it tho.

Tupson444, 2023-11-14:
Oh I missed that line saying it's fixed, good to know :) I didn't want to experiment with alpha fw yet, as I just got my new printer.

closed 2024-06-25 by github-actions[bot]