prusa3d/Prusa-Firmware-Buddy #3271: [QUESTION] Change Input Shaper values not possible.

rainer1146, 2023-09-18:
Hello, why is a change input shaper values even in the final version 5.0 not possible?
The menu exists but it is grayed out and cannot be changed.
I have determined my values with a test print.
If I give them via the start Gcode the printer executes these or the preset values of the firmware?

marek26340, 2023-09-18:
Why not just edit the relevant lines in the start Gcode in the slicer? Even if you'd edit these on the printer, PrusaSlicer will, by default, change the values a couple of times on different layer heights using Gcode.

rainer1146, 2023-09-18:
I'm already doing that. 
What sense does the menu in the firmware make if you can't set anything?
In Prusa Slicer there is no Input Shaper entry in the start code anymore. Therefore, I thought that it is now controlled via the firmware.
Then the gcode has priority over the firmware settings when I change the frequencies in the start gcode?

danopernis, 2024-05-30:
@rainer1146 We were debating if we should include "unfinished" menu or hide it. We eventually decided that it is still valuable to see the values, even if the menu doesn't allow changing them yet.

In the meantime, you can edit values as suggested by @marek26340

closed 2024-05-30 by danopernis