prusa3d/Prusa-Firmware-Buddy #3272: [BUG] <Enter comprehensive title>

geekcouplettv, 2023-09-18:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MK4]

**Printer firmware version** - [5.0.0]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Self test failing]

**USB drive or USB/Octoprint**
  Please let us know if you print via USB and Prusa connect

**Describe the bug**
Firmware is working. But I have bug error on my two mk4 printers failing X assis test's. And its not even starting the Y and Z axxis test.
Earlier on one printer witch was assembled week ago. All test's was passing on 4.7.2 firmware but after update to 5.0.0 is failing X axxis test

**How to reproduce**
Run Self Test
**Expected behavior**
  X axxis is crushing to the left many times and failes

**G-code**
none

**Crash dump file**
  none

**Video**
  I can send it

Delita232, 2023-09-19:
I have a similar issue but with my y axis. Prints are coming out fine though.

kgpsyched, 2023-09-19:
Same issue upon upgrading to 5.0 with my MK4.  X-axis failed two consecutive times in self test with LOTS of bumping and motor grinding sounds.  Passed the third and fourth time I ran the self-test, and no problems since.  Reported to Prusa help chat who had me run one additional self-test to ensure all is working properly now (passed again).

Silvenga, 2023-09-22:
Same, issues only after upgrading to 5.0 - going from no issues with auto-homing to failures 90% of the time.

Silvenga, 2023-09-27:
So after messing with the printer - I can get 100% success-rate when holding the printer during the homing - it appears the rocking caused by the (somewhat more aggressive) homing was causing the test to fail.

(and to be clear, I did not mess with the belts, since the belts were factory tuned only a month or so ago - 
 I didn't want my own inability to follow tensioning directions to impact my theories)

Silvenga, 2023-11-07:
Wanted to say that 5.0.1 resolved my issues here.

> As part of the self-test changes, we have also implemented an improved algorithm for calibrating homing sensitivity.

feutl, 2023-11-28:
I thought 5.0.1 or 5.1 resolved it, but it just came back after a while.
Every second print does not start because of the x-axis test at the beginning of the print, which is pretty annoying.

Mikyner, 2024-01-07:
The same problem haunts me to this day. 90% of prints start with constant X bumps, Calibration, recalibration etc.
The OS XY check always passes without issue.

Sebastian1989101, 2024-01-28:
Same here... how can something like this not be fixed asap? https://youtu.be/NDHEbcZWi1U

closed 2024-06-25 by github-actions[bot]