prusa3d/Prusa-Firmware-Buddy #3273: [BUG] 5.1.0-alpha Prusa Mini fails to home X axis

brettvitaz, 2023-09-18:
**Printer type** - Mini

**Printer firmware version** - 5.1.0-alpha

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint**
Printing via USB

**Describe the bug**
Issuing a Home command from the Control menu, or trying to print anything, causes the X axis to move left slightly, then all the way to the extreme right, then it moves back and forth crashing into the extreme right continually until the printer resets.

**How to reproduce**
Select "Control", "Home"

**Expected behavior**
Successful homing, less crashing.

**Video**
https://github.com/prusa3d/Prusa-Firmware-Buddy/assets/7563757/b07f8854-f30f-4d61-a7f9-c7505818c726

kevman28, 2023-09-19:
Same exact problem here as well.  Have an identical video.

kevman28, 2023-09-19:
I might have figured this out. I unplugged my octopi from the USB port and the auto home worked without crashing the hotend.  Not sure if you have the same setup.

brettvitaz, 2023-09-19:
I do not have octopi connected.

tomasjakubik, 2023-09-19:
Hi, just a few prerequisites. Please check that it doesn't hit anything soft, like fan cable or the wire harness.
After that, can you get log from the USB console? You would need USB cable and some program to read serial console. That depends on OS and everything. It would look like this:
```
echo:enqueueing "G28"
echo:endstops hit:  X:13.37
echo:busy: processing
echo:endstops hit:  X:10.01
```
If the second hit is more than 0.2 away from 10, then it will try again.

Gyrxiur, 2023-09-19:
For me it helped to tighten x belt a little bit.

Larswa, 2023-09-19:
Same behavior. Xbelt is as tight as I am comfortable tightening it. It plays bass notes when I flick it.

Gyrxiur, 2023-09-19:
Have you tried prusa belt tuner? maybe it is too tight?

Stigern, 2023-09-19:
Same behavior, happens every time.

brettvitaz, 2023-09-19:
Here is the response in the serial console:

```
>>> G28
SENDING:G28
echo:endstops hit:  X:91.06
echo:endstops hit:  X:10.64
echo:endstops hit:  X:10.64
echo:endstops hit:  X:10.64
echo:endstops hit:  X:10.64
echo:endstops hit:  X:10.64
echo:endstops hit:  X:10.64
echo:endstops hit:  X:10.64
echo:endstops hit:  X:10.64
echo:endstops hit:  X:10.64
echo:endstops hit:  X:10.64
echo:endstops hit:  X:10.64
echo:endstops hit:  X:10.64
echo:endstops hit:  X:10.64
echo:endstops hit:  X:10.64
echo:endstops hit:  X:10.64
echo:endstops hit:  X:10.64
echo:endstops hit:  X:10.64
echo:endstops hit:  X:10.64
Can't read from printer (disconnected?) (SerialException): read failed: [Errno 6] Device not configured
```

tomasjakubik, 2023-09-19:
> Here is the response in the serial console:
Thanks. It is consistent, but different than it should be. No promises, but I will have a look.

Larswa, 2023-09-19:
> Have you tried prusa belt tuner? maybe it is too tight?

Didn't know about the tuner. Clever little use of a phone. I loosened belts a quarter of a turn, to satisfy it, but the same result. It cant successfully home

brettvitaz, 2023-09-19:
After adjusting the X and Y belt tension, I was able to get the printer to home:

```
>>> G28
SENDING:G28
echo:endstops hit:  X:13.99
echo:endstops hit:  X:10.01
echo:endstops hit:  Y:-37.10
echo:endstops hit:  Y:-10.00
echo:endstops hit:  Z:-2.03
echo:endstops hit:  Z:-2.00
```

tomasjakubik, 2023-09-19:
It won't fix the problem, but could some of you that still cannot home try going slightly slower?
```
G0 F2500
G28 N XY
```
I think it skips a step on each bump, so contrary to my advice it would help if it have hit something soft. We may have changed some seemingly unrelated homing stuff, so it skips more often than before.

Larswa, 2023-09-19:
A Factory Reset with reinstall of firmware and out of the box start calibration fixed it for me. For starters I played with the tightness of the belts, but after it was ok according to the prusa calibrator using my phone, I still had the issue.  The firmware reset seemed to clear it for me.

Larswa, 2023-09-19:
> A Factory Reset with reinstall of firmware and out of the box start calibration fixed it for me. For starters I played with the tightness of the belts, but after it was ok according to the prusa calibrator using my phone, I still had the issue. The firmware reset seemed to clear it for me.

DAMMIT ...  that only lasted one print ... the bonky benchy.  After putting my printer back and hooking it up to prusa connect again, it failed homing the x axis again.  The belts are smack in the center of the prusa tuning by sound app.

kevman28, 2023-09-19:
> G28 N XY

Send: G0 F2500
Recv: ok
Recv:  T:23.06/0.00 B:23.20/0.00 A:28.20/0.00 @:0 B@:0
Recv:  T:23.00/0.00 B:23.28/0.00 A:28.23/0.00 @:0 B@:0
Recv:  T:22.97/0.00 B:23.32/0.00 A:28.31/0.00 @:0 B@:0
Recv:  T:23.06/0.00 B:23.20/0.00 A:28.31/0.00 @:0 B@:0
Recv:  T:23.00/0.00 B:23.28/0.00 A:28.33/0.00 @:0 B@:0
Send: G28 N XY
Recv:  T:23.03/0.00 B:23.32/0.00 A:28.39/0.00 @:0 B@:0
Recv: echo:endstops hit:  X:10.87
Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
Recv: echo:endstops hit:  X:10.64
Recv:  T:23.03/0.00 B:23.12/0.00 A:28.33/0.00 @:0 B@:0
Recv: echo:endstops hit:  X:10.64
Recv: echo:endstops hit:  X:10.64
Recv: echo:busy: processing
Recv:  T:23.00/0.00 B:23.16/0.00 A:28.44/0.00 @:0 B@:0
Recv: echo:endstops hit:  X:10.64
Recv: echo:busy: processing
Recv: echo:endstops hit:  X:10.64
Recv:  T:22.97/0.00 B:23.24/0.00 A:28.44/0.00 @:0 B@:0
Recv: echo:endstops hit:  X:10.64
Recv: echo:endstops hit:  X:10.64
Recv: echo:busy: processing
Recv: echo:endstops hit:  X:10.64
Recv:  T:22.97/0.00 B:23.24/0.00 A:28.44/0.00 @:0 B@:0
Recv: echo:endstops hit:  X:10.64
Recv: echo:endstops hit:  X:10.64
Recv: echo:busy: processing
Recv: echo:endstops hit:  X:10.64
Recv:  T:22.97/0.00 B:23.24/0.00 A:28.44/0.00 @:0 B@:0
Recv: echo:endstops hit:  X:10.50
Recv: echo:busy: processing
Recv: echo:endstops hit:  X:10.14
Recv:  T:23.03/0.00 B:23.28/0.00 A:28.46/0.00 @:0 B@:0
Recv: echo:endstops hit:  X:10.50
Recv: echo:endstops hit:  X:10.14
Recv: echo:busy: processing
Recv: echo:endstops hit:  X:10.64
Recv:  T:22.97/0.00 B:23.32/0.00 A:28.54/0.00 @:0 B@:0
Recv: echo:busy: processing

tomasjakubik, 2023-09-19:
@kevman28 Thanks.
Ok, so it is not too fast. Not sure what it could be yet. My printer needs around F5000 to start doing the same thing.

rosenstand, 2023-09-19:
> After adjusting the X and Y belt tension, I was able to get the printer to home:
> 
> ```
> >>> G28
> SENDING:G28
> echo:endstops hit:  X:13.99
> echo:endstops hit:  X:10.01
> echo:endstops hit:  Y:-37.10
> echo:endstops hit:  Y:-10.00
> echo:endstops hit:  Z:-2.03
> echo:endstops hit:  Z:-2.00
> ```

Did you tighten or loosen them? I have mine set right in the middle of the Prusament GT2 belt tension tool (https://www.printables.com/model/46639-tension-meter-for-the-gt2-belts-of-i3-mk3s-or-prus) but it happens here too. Never had problems with the X-axis on 4.4.1 (or earlier).

murk-sy, 2023-09-19:
Having the same issue on one of the Minis, but only on Y. Haven't tested all of them yet.

During calibration it seemed to work fine (bumping maybe 3 times?) but after doing a normal print it failed. Belt tension checked with belt tuner.

Will try to pull some data in the evening.

tomasjakubik, 2023-09-19:
If you still feel adventurous, you can try increasing stall sensitivity.
```
M914 I1 X140
G28 X
```
(or `Y`)
Apart from that, I still have nothing.

dahuafschmied, 2023-09-19:
M914 X140 helps

echo:endstops hit:  Z:3.99
echo:endstops hit:  X:12.07
echo:busy: processing
echo:endstops hit:  X:10.64
echo:endstops hit:  X:10.50
echo:endstops hit:  X:10.14
X:180.40 Y:-3.00 Z:4.00 E:0.00 Count X:18040 Y:-300 Z:1600
ok

tomasjakubik, 2023-09-19:
Even the first try bumps more times than it should and homing precision won't be quite top. So this is not the solution either :thinking:.

kevman28, 2023-09-19:
> G28 X

Worked on X, but failed when  I tried to start the print.

murk-sy, 2023-09-19:
```
>>> m914 y140
SENDING:M914 Y140
>>> g28
SENDING:G28
echo:endstops hit:  X:10.01
echo:endstops hit:  Y:-10.48
echo:endstops hit:  Y:-10.01
echo:endstops hit:  Z:-59.81
echo:endstops hit:  Z:-2.00
>>> m914 y130
SENDING:M914 Y130
>>> g28
SENDING:G28
echo:endstops hit:  X:13.99
echo:endstops hit:  X:10.00
echo:endstops hit:  Y:-37.23
echo:endstops hit:  Y:-10.65
echo:endstops hit:  Y:-10.64
echo:endstops hit:  Y:-10.63
echo:endstops hit:  Y:-10.65
echo:endstops hit:  Y:-10.65
echo:endstops hit:  Y:-10.63
echo:endstops hit:  Y:-10.64
echo:endstops hit:  Y:-10.64
echo:endstops hit:  Y:-10.64
echo:endstops hit:  Y:-10.64
echo:endstops hit:  Y:-10.64
echo:endstops hit:  Y:-10.64
echo:endstops hit:  Y:-10.64
echo:endstops hit:  Y:-10.65
echo:endstops hit:  Y:-10.63
echo:endstops hit:  Y:-10.64
echo:endstops hit:  Y:-10.64
```

Works with 140, fails with 130 on mini 2. Will test others as well.

dahuafschmied, 2023-09-19:
My problem was the belt tension. is was at the lower end of the belt tool. changed it to the upper end of the scale. works fine now. hits 2x the endstop and ok.

brettvitaz, 2023-09-19:
> Did you tighten or loosen them? I have mine set right in the middle of the Prusament GT2 belt tension tool (https://www.printables.com/model/46639-tension-meter-for-the-gt2-belts-of-i3-mk3s-or-prus) but it happens here too. Never had problems with the X-axis on 4.4.1 (or earlier).

I had to adjust both X and Y. I tightened the belts by half turn increments until the G28 response came back OK and homing succeeded. It was all-or-nothing for me and tightening too much made the problem come back. Firmware v4.4.1 never gave me any trouble, but I can imagine that much stricter parameters are necessary to enure the proper tuning for input shaping. (BTW, my "bonkers" speed benchy results were surprisingly good.)

kevd, 2023-09-19:
@brettvitaz - that worked for me too thanks!

samgonemudding, 2023-09-20:
Also adjusted to upper end of belt test and started working

jbienz, 2023-09-20:
For as serious as this error seems (watchdog triggered reboot followed by red screen and alarm), I was surprised to see it fixed by a belt tension too. I also went for the higher end of the tension tool.

Might want to change the severity of this error, and / or add information about tensioning the belt.

morrispe65, 2023-09-21:
Was getting X axis homing crashes on my Mini as well. Tightened the belts with the phone app tensioner and working now. This was one of the original pre-orders with the quick assembly and had never adjusted the belt tension since.

rebelsdigital, 2023-09-22:
Same for as well, exact problem here. Have an identical video.

DalbyTech, 2023-09-28:
Same problem for me too! Downgrading the firmware back to 'prusa3d_fw_4_4_1_MINI.zip' fixed it.
Here's a video of the behaviour: [https://drive.google.com/file/d/1yfZJhwwaIlvfg5iY9aqxqUQ5g8vQbwJK/view?usp=sharing]

benoitcosials, 2023-10-01:
I got the problem as soon as I updated. My X belt was very loos, thus the stepper motor power make the belt slip with a characteristic noise (like "noke"). After tension the belt until the head of the screw went flat with the plastic part, I pass through the homing test procedure. No slip anymore.

KBalage, 2023-10-03:
An added C-ring can help to make homing a success:
https://www.printables.com/model/600173-prusa-mini-is-alpha-firmware-homing-error-workarou

thexperiments, 2023-10-22:
Same problem here, tried with tighter belts and right belt tension.
Now checking if the c-rings help. However I was only able to print them by wedging a flat piece of metal in the place where the printhead smacks into the rest of the assembly. (thus I also suspect they will work)

plaes, 2023-10-23:
Issue still seems to be present in 5.1.0-alpha2.

MihRei, 2023-10-29:
It happend to me too but after i made the x Belt more tight it stopped happening

Daddy-Boomalati, 2023-10-29:
I am having the exact same issue. I I tried a factory reset and it homed okay one time, and then continued to bang away at the X home. I downgraded to 4.4.1 and the printer worked fine. I reinstalled alpha 2 and it immediately crashed again.

There is nothing between the carriage and the frame.

Daddy-Boomalati, 2023-11-06:
I wanted to give an update. I used the Mini tuning guide on the Prusa3D help tab and after printing and using the belt tensioning tool, it works fine. I didn’t know the tool existed. FYI, I printed it from generic Amazon PLA and it worked fine.

Daddy-Boomalati, 2023-11-07:
> > I wanted to give an update. I used the Mini tuning guide on the Prusa3D help tab and after printing and using the belt tensioning tool, it works fine. I didn’t know the tool existed. FYI, I printed it from generic Amazon PLA and it worked fine.
> 
> Because PLA is much harder than PETG, the tool will not indicate the otherwise recommended tension. Instead your belts will be tighter, which is what others are also reporting to be needed for the input shaper homing to succeed on the Mini.
> 
> At least for me, the belts were too loose when using the tool printed correctly (using Prusament PETG). That was on alpha 1 though.

Sorry, I mispoke. I meant PETG. Good info for anyone reading the thread though, so thank you!

murk-sy, 2023-11-09:
As a side note: Prusa has an official belt tuning website which you can access on your phone. It supports Mini since a few weeks ago.
https://belt.connect.prusa3d.com/

tomasjakubik, 2023-11-09:
So the difference in homing between 4.4.1 and 5.1 is that MINI tried to do the second bump. That second bump is meant for MK4 and XL where it is necessary for toolchanger, crash detections and so on. MINI is built differently and doesn't need to home so precisely, so the next release will return back to the single hit and done.

If you can tune the belts, that is always a good idea.
If those C rings work for you, it is an option. It changes the hit position, the motor is in a different phase and the possibility to skip a step changes. But you could just wait for the next release that will come soon-ish:tm:.

Daddy-Boomalati, 2023-11-09:
Thank you for your work on the project. It is working great now, so I’ll just roll with it.

closed 2023-11-09 by brettvitaz