prusa3d/Prusa-Firmware-Buddy #3274: [BUG] Failing to home Y axis

wordym, 2023-09-19:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** -  5.1.0-alpha
**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint**
  Printing via USB

**Describe the bug**
 Selecting the Home command from the Control menu or trying to print anything causes the Y axis to move slightly to the front, then to the very back and then it moves back and forth crashing into the back continuously until the printer resets.

**How to reproduce**
  Via menu, Control -> AutoHome

**Expected behavior**
  Successfull homing without crashing as it was on 4.4.1

**G-code**
  [BonkersBenchy_PLA_MINI_IS_16m.gcode] (https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/download/v5.1.0-alpha1/BonkersBenchy_PLA_MINI_IS_16m.gcode)

**Video**
(https://github.com/prusa3d/Prusa-Firmware-Buddy/assets/110923281/ecbbbb60-6039-4696-b542-879961eca3cc)

Gyrxiur, 2023-09-19:
For me it helped to tighten x belt a little bit.

Netfreak25, 2023-09-19:
I also had a issue with Y-homing when running the diagnostics.
I uncabled the mini and placed it minimal to another possition and suddenly it was working again

wordym, 2023-09-19:
> For me it helped to tighten x belt a little bit.

In my case it really was the tightness of the Y belt - looks like this firmware is way more sensitive to it than 4.4.1. The Y-belt was tight enough according to the older Prusament PETG gauge (https://www.printables.com/model/46639-tension-meter-for-the-gt2-belts-of-i3-mk3s-or-prus), but according to the app it wasn't tight enough. Tightened it to the belt tuning app and it worked.

SmileyNY85, 2023-09-19:
Having the same issue too. Reverted back to 4.4.1 and it's working fine. Flash back to this firmware and back to having issues with the Y. I have cleaned and lubricated the rods but hasn't help.

MoritzLoebmann, 2023-09-20:
Same problem! 
But tightening the belts a little helped. Belt tuner app says it's a bit to tight now.

zatekk, 2023-09-20:
Same problem. Tightening of the Y belt helped.

commanderf, 2023-09-22:
Thanks for the tip with tightening the belt: Fixed my problem too

tbeesto-git, 2023-09-25:
Tightened the belt slightly and it homed ok once, next time it failed again. Tightened again and no joy, then at quarter turns testing each time until I can tighten no more, all failed.

Loosened all the way and started at the beginning, each time it fails no matter how tight then belt is so seems very sensitive. 
 
Flashed back to previous firmware for now and all working fine.

MoritzLoebmann, 2023-09-25:
> Tightened the belt slightly and it homed ok once, next time it failed again. Tightened again and no joy, then at quarter turns testing each time until I can tighten no more, all failed.
>
> Loosened [...]

This seems to be another solution:
https://forum.prusa3d.com/forum/input-shaper-mini/5-1-0-alpha1-firmware-for-original-prusa-mini-mini-with-input-shaper-and-prusa-connect/paged/2/#post-674594

Prusa-Support, 2023-12-28:
As per the [FW 5.1.0-alpha1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0-alpha1) release notes
> In rare cases the printer might have issues with homing, a temporary fix is to hold the print head for a while while homing.

I suspect something similar was occurring on the Y-axis test but not reproduced by our developers before the release.
I'd like to know if the problem persists on the final and most recent release [FW 5.1.2](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.2).

Michele Moramarco
Prusa Research

Prusa-Support, 2024-01-24:
To our understanding, the problem is actually not reproducible on the most recent firmware release.
Please let us close this issue.

Michele Moramarco
Prusa Research

closed 2024-01-24 by Prusa-Support