prusa3d/Prusa-Firmware-Buddy #3275: [BUG] BSOD when trying to print anything

Gyrxiur, 2023-09-19:
**Printer type** - [MINI]

**Printer firmware version** - [5.1.0-alpha1]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**USB drive**
  Please let us know if you print via USB flash drive or USB/Octoprint

**Describe the bug**
  When I try to print any file, with or without input shaper, when I open it in printer menu, I get BSOD, also I get BSOD when trying to print it through PrusaLink.

**How to reproduce**
  Update firmware to 5.1.0-alpha1 and try to print anything.

**Expected behavior**
  It should open up file preview in printer and allow me to print or be able to print through PrusaLink.

**G-code**
  Not needed, it does with every gcode older firmware had no problems, or with new with fully updated Prusaslicer.

**Crash dump file**
  https://ulozto.net/file/myRkfsbwy4zV/dump-buddy-bin

Gyrxiur, 2023-09-19:
Ok, so I have found out, it has something to do with a lot of files on flash drive, if I have only few, it works.

Gyrxiur, 2023-09-19:
And also, it likes to crash when connecting flash drive, better to turn it off, plug it in and turn on.

LukasLendvorsky, 2023-09-19:
@Gyrxiur Can you please attach gcode you are trying to print?

LukasLendvorsky, 2023-09-19:
@Gyrxiur Can you please attach gcode you are trying to print?

Sorry about closing this issue, this it was missclick

murk-sy, 2023-09-19:
The BSOD is on `task:displayTask?`
[dump_buddy.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12665153/dump_buddy.zip)

Removing all other files and generating gcode without previews does not fix the issue.

Reformatting the USB drive and then putting all gcode files back fixes the issue.

It's probably related to long term USB flash drive data deterioration that I've dealt with many many times, see https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1793

Gyrxiur, 2023-09-19:
Yes, task:displayTask.
I don't think it is deterioration, on previous firmware files work ok, even oldest ones. It has something to do with many files on drive, I have over 400 files on it, and also with speed, on faster drive, if I only have few files, they work ok, but if I copy all of the files, even the ones, previously working, no longer work. And on that slower drive, even if I keep only few files, some of them load, some don't, randomly. Weird. But as I said, with 4.4.1 same files work.

Gyrxiur, 2023-09-19:
> @Gyrxiur Can you please attach gcode you are trying to print?
> 
> Sorry about closing this issue, this it was missclick

It doesn't matter what file it is. Even that IS benchy doesn't open if I have a lot of files on drive. Also, I have BSOD even when I connect drive with a lot of files, and I have tried reformatting it and copying same files to it again. Drive is ok, works fine with 4.4.1 fw. There is probably some issue when I have a lot of files on flashdrive, which is worse on slower drive.

calrock27, 2023-09-19:
I have the same issue as well. Newly flashed 5.0.1 alpha1 on my Mini+. Every Gcode file generates a crash, even previously known working files and the presliced benchy files in the 5.0.1 alpha1.

BSOD code: task:displayTask?

Dump:
[dump_buddy.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12666610/dump_buddy.zip)

This is utilizing an 11 month old Sandisk 64GB flash drive with approx. 20 sliced files on the drive. https://www.amazon.com/gp/product/B01EZ0X3L8/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1

calrock27, 2023-09-19:
> I have the same issue as well. Newly flashed 5.0.1 alpha1 on my Mini+. Every Gcode file generates a crash, even previously known working files and the presliced benchy files in the 5.0.1 alpha1. [...]

Interestingly enough - I now have a print that is able to be started. I reformatted the drive using a fat32 specific tool for larger drives and tried again. https://pendriveapps.com/format-fat32-larger-than-32gb/. Does anyone else care to try? Perhaps this is a drive format issue?

jwidess, 2023-09-20:
Had the same problem, BSOD with USB drive that had ~100 Gcode files. Tried deleting them all and still had a crash with just the single newly sliced file. I had to reformat my 16GB flash drive for it to work without BSOD.

Dump from my final crash with only the single newly created Gcode file, before formatting.
[dump_buddy-JdogAwesome.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12668983/dump_buddy-JdogAwesome.zip)
[image]

DFEvans, 2023-09-21:
I have found that this issue seems to be at least sometimes caused by the "Format of G-code thumbnails" option for my Minis. Changing it to the new "QOI" option and setting the G-code thumbnails option to `16x16, 220x124, 200x240` (per the new presets for IS printing) seems to be a lot more reliable. PNG thumbnail prints, the old default, are troublesome.

I'm not sure if this applies to printers other than the Mini, though - did QOI thumbnails get adopted for those, too?

dpenezic, 2023-09-22:
I have the same issue, however, I did some more tests and found the following: 
* if you plug in USB stick with a large list of files, the printer will wait for 2-5 s, and BSOD
* if you are not in the main menu, sometimes you will get BSOD, but sometimes won't (it looks like firmware in the background trying to do something with the USB drive)
*  after reboot you are able to see a list of files and go a cross, however if you select one you may or not get BSOD
* if I select old files in a directory with few files, I get information about gcode, but without preview, if select to print BSOD happens
* if i select on long list files (the oldest one) same happens as before, however on top of the list (new one) BSOD happens without showing file information or a preview

Altogether it is impossible to print anything, and I use an original Prusa USB stick.

It is look like some issue with FET stack and manipulating with files ... maybe some buffer overflow

murk-sy, 2023-09-22:
@dpenezic Can you try doing what I did? Copy files off the drive, reformat it and then copy all the files back. Afterwards it worked perfectly fine for me.

I've long suspected the filesystem to be the cause of a lot of my woes over the years and I feel this is just another one of them.

DFEvans, 2023-09-22:
Further to my above comment, I've now hit this with 4-5 QOI files, so either the thumbnails are barely related, or it was simply chance. I've downgraded my firmware again for the time being, since there's no clear workaround other than cleaning the USB every couple of prints.

dpenezic, 2023-09-22:
@murk-sy I did try with a completely new from-box USB stick and with only one test file (benchy 16m), without success. What may be an issue is fact that I do everything on Linux , and don't have any MS Win comp near to try format USB stick.

Additionally, I don't see a direct connection, because I have some experience with micro/midi controllers and most of the time is a work or not-work state of reading ... in this case, because of optimization, it is possible that some stack/buffer overflow happened .

vinscarlata, 2023-10-12:
+1 for BSOD with 400+ gcode files. Mine is properly FAT32 formatted (using rufus). I think someone asked that. 

In my case, after the BSOD started happening with my main USB stick, I restoring the 4.4.1 firmware and was able to print fine from that USB stick again. So whatever is wrong with the USB stick, it only bothers the alpha firmware, not 4.4.1.

Found this bug report. I reinstalled the alpha firmware, deleted all the extra files from the USB stick and it did not fix it, but I did notice that a bunch of the gcode files that were previously transferred via wifi by Prusa Slicer had jacked up timestamps in 2097. 

Copied a failing gcode to another USB stick and it worked fine. Re-formatted the problematic USB stick, and I can print from it fine as well.

michalmiddleton, 2023-10-15:
I had this happen as well. BSOD popped up during (or near the end) of upload via PrusaLink. 
Using the original USB drive with about 417 files on it. I just retried and it worked the next time. 
Crash dump attached.
[dump_buddy.bin.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12910494/dump_buddy.bin.zip)

Not sure if it matters but I am also affected by https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3156 , however I am now using the original USB drive and no longer suffer from the data corruption.

closed 2024-06-25 by github-actions[bot]