prusa3d/Prusa-Firmware-Buddy #3276: [BUG] Unload distance seems slightly shorter on MINI 5.1.0 alpha

darraghbr, 2023-09-19:
**Printer type** - MINI

**Printer firmware version** - 5.1.0 alpha

**Original or Custom firmware** - Original

**Optional upgrades** - Filament runout sensor, Bondtech extruder, Bondtech PTFE fittings

**USB drive or USB/Octoprint** - USB drive

**Describe the bug**

I have installed the Bondtech PTFE fittings, this enables the use of a single section of PTFE that goes all the way from the extruder down into the heatbreak. The section of PTFE that comes with the fittings is maybe slightly longer than it really needs to be, but it was working fine as is. 

Before 5.1.0 alpha this was working fine, I always needed to press the 'MORE' button when feeding filament to get a nice purge but that never bothered me as I did it before anyway. On unload it would unload fine and completely clear the extruder before the extruder shut off.

After 5.1.0 alpha the unload distance appears to be slightly shorter? It now no longer clears the extruder before it shuts off. 

**How to reproduce**

- Install Bondtech PTFE fittings with PTFE tube that accompanies it.
- Unload filament

**Expected behavior**

The filament should unload cleanly.
 - By running the unload until the filament sensor no longer detects filament (not a great option as the filament sensor is optional)
 - Have the unload distance be set as a parameter in the same menu where you can change the extruder direction and steps per mm (Most elegant, most adaptable)
 - Having a 'more' button on the unload screen similar to the load screen (Might be the quick and dirty option?)

schleifi, 2024-01-03:
It's already a pre-5.1.0 problem.

The "length" of the unload should be adjustable by a "firmware"-parameter.

The filament-sensor does not work as (at least the original Prusa one) is far "behind" the extruder.

danopernis, 2024-05-30:
@dstulken or anybody else, please confirm if this happens also on stock Prusa MINI. If it does, this might get a larger priority for us and we will have to dig into code history to see what the original value was. If it was indeed changed by accident, we will fix it in some of the next releases.

dstulken, 2024-05-30:
Yes, my Minis are 100% stock.   For a while, I had a pair of them side by side, one with a 4.x firmware, and one on the 5.x firmware, and the unloading behavior was definitely different between them.

overlander, 2024-07-16:
I've been having this issue on a stock non plus mini, it started around fw v5. I've measured my Bowden and its 178.6 mm, I'm guessing that's the problem, the store says it should be 175. it would be ideal if this was just a user adjustable setting.