prusa3d/Prusa-Firmware-Buddy #3277: [BUG] Flashing screen on MINI 5.1.0 alpha

Erik262, 2023-09-19:
**Printer type** - MINI

**Printer firmware version** - original

**Original or Custom firmware** -  5.1.0 alpha

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint**
  USB and also same happens via PrusaConnect

**Describe the bug**
  Right after the Warning message about Alpha Firmware, the whole screen keeps refreshing every 3 second

**How to reproduce**
  This error happened right after flashing the alpha firmware

**Expected behavior**
  No permanent flashing screen UI

**G-code**
 Not necessary 

**Crash dump file**
  The printer keeps printing fine without any issues.

**Video**
https://github.com/prusa3d/Prusa-Firmware-Buddy/assets/11977647/d3cedbca-3a62-4501-be4d-6dd3035e3af1

diminDDL, 2023-09-19:
I have the exact same issue. I believe it's a signal integrity issue, as I have scoped out the data lines during the bootloader/alpha firmware warning screens (when this issue is not present) and then the main home screen.

# Boot screen signals:
[image]

# Boot screen signals, zoomed in:
[image]

# Home screen signals:
[image]

# Home screen signals, zoomed in:
[image]

# Home screen signals, zoomed in (different section):
[image]

As can be seen above, the display updates during the boot screen phase all have pretty good edges no overshoots and in general you can clearly see a digital signal there. The home screen updates show a very ringy and triangular looking signal. So it appears that during an effort to either increase the screen refresh rate or while porting from a different display, some clocking settings have been set slightly too high where it works fine on some machine, but breaks on certain others.

D3nkis, 2023-09-19:
Hi folks, the issue is same.

PhilippeGalid, 2023-09-19:
Hi after Update to MINI 5.1.0 alpha Firmware i have the same problem, Flashing Screen, i go now back to the old Firmware

R4ynor, 2023-09-20:
I have the same problem, but in my case when a motor is running I can here it glitching. The worst case is the extruder motor:

https://github.com/prusa3d/Prusa-Firmware-Buddy/assets/9499061/5cb1af4c-beb1-4422-be97-9e76044ef2ae

Lino77, 2023-09-20:
I have this problem only on the display, 3.2" and on the 2.4"from Fysetec. On the 2 original from Prusa LDO A01 and A03 that I have and one from Bigtreetech I do not have this problem.

CuriousTimo, 2023-09-20:
So @R4ynor seems to be running an original Prusa Mini.
But are the others running an original printer or a Fysetc clone? And if it's a clone, what screen? The 3.2" or 2.4"?

fabry-68, 2023-09-20:
I have the same problem on 2 x Fysetc clones with 3.2" LCD.

ZeekWorld, 2023-09-20:
Same, Fysetc 3.2".

dje33, 2023-09-20:
Same, Fystec 2.8"

Ferrugius, 2023-09-21:
Fysetc 3.2" here.

xXNoceboXx, 2023-09-22:
I have the same issue with Fystec 2.8 screen 

https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3301#issue-1908538040

nolookjones, 2023-09-29:
same issue with a Fysetc 2.8 screen

Gabri988, 2023-10-01:
Same problem...Fysetc 3.2 lcd

EnnjiBennji, 2023-10-07:
same issue with a Fysetc 2.8 screen

g8toparadise, 2023-10-09:
Any news from Prusa on this issue ? Since it seems to happen mainly on clone do you think we can  hope for a correction ?

SteynGuelen, 2023-10-11:
Same issue here with 2.8 Fysetc screen.

diminDDL, 2023-10-12:
> Any news from Prusa on this issue ? Since it seems to happen mainly on clone do you think we can hope for a correction ?

Prusa does not actually develop in this repository, this appears to be just a mirror they push to every major release, so we will likely see/hear about it after the next release.

CuriousTimo, 2023-10-16:
As a quick test, I have ordered a new LCD screen made by Bigtreetech.

I can confirm that swapping over only the LCD from a Fysetc one to a Bigtreetech fixes the issue.
Everything else including the ribbon cable was left in place.

So the flashing issue seems to be with the Fysetc screens.

I doubt that Prusa will fix a clone hardware-related issue. So in my personal situation, this will be my permanent fix.

KR4DIO, 2023-10-16:
> I can confirm that swapping over only the LCD from a Fysetc one to a Bigtreetech fixes the issue. Everything else including the ribbon cable was left in place.

Did you swap the LCD panel onto the fysetc board? Or swap the whole assembly? If the latter, are there any obvious differences in the components between the pin header and FPC connector?

g8toparadise, 2023-10-16:
@CuriousTimo 
Thanks for the head up !!!! 

@ZeekWorld 
https://fr.aliexpress.com/item/4001267632446.html?spm=a2g0o.productlist.main.21.3134dT0MdT0MHB&algo_pvid=fd49074d-0f17-481f-a8d9-02704fd9cf50&algo_exp_id=fd49074d-0f17-481f-a8d9-02704fd9cf50-10&pdp_npi=4%40dis%21EUR%2119.11%2112.8%21%21%2119.64%21%21%40211b440316974778766633708e93f8%2110000015531261574%21sea%21FR%21731355889%21&curPageLogUid=eAsfLLECGhkn

CuriousTimo, 2023-10-16:
> Did you swap the LCD panel onto the fysetc board? Or swap the whole assembly? If the latter, are there any obvious differences in the components between the pin header and FPC connector?

I only swapped the LCD. Nothing else. I will have to check if I can find any obvious differences between the two.

@g8toparadise Yes it is item 4001267632446.

PhilippeGalid, 2023-10-16:
Did we have to buy the bigtreetech Display to fix the Problem on the Prusa Mini Clone, or you think Prusa will fix the Problem with a New Firmeware for the Display, did we have to what or buy now a new Display is cost 20euro.

https://biqu.equipment/products/clone-prusa-mini-buddy-control-board-integrated-tmc2209-driver-mini-lcd28-v1-0-switch-3d-printer-parts-for-prusa-mini-printer

R4ynor, 2023-10-16:
In my case I put this Bigtreetech display in my original Prusa as a replacement and couses the flashing problem. I'll try tomorrow to swap the LCD in the original board with this one like @CuriousTimo.

R4ynor, 2023-10-17:
I can confirm that the problem is related to the PCB and not with the LCD from bigtreetech, my printer now works flawlessly.

CuriousTimo, 2023-10-20:
Update with bad news:

My BTT LCD worked flawlessly for about three days. Last night I noticed it started flashing.

- I leave my mini on 24/7
- Tried a restart
- Tried going back to the Fysetc LCD

It keeps flashing now every time.

Unfortunately, I do not have the equipment to do any further in-depth measuring. But I think that @diminDDL is on the right track that in the newer firmware something has changed causing issues on the data lines and clock signal.

The clone LCD screen seem to be less tolerating.

PhilippeGalid, 2023-10-20:
There is a New Fireware out now did the Fysetec LCD problem fix it now with the New Firmware ? Did some tested ?

KR4DIO, 2023-10-20:
> There is a New Fireware out now did the Fysetec LCD problem fix it now with the New Firmware ? Did some tested ?

No change here with the Fysetc 3.2 LCD

RasmusLauritsen, 2023-10-20:
> > There is a New Fireware out now did the Fysetec LCD problem fix it now with the New Firmware ? Did some tested ?
> 
> No change here with the Fysetc 3.2 LCD

It's  the same for me. Still flashing the Fysetc 3.2 LCD.

PhilippeGalid, 2023-10-20:
I'm not a programmer but my idea is perhaps if you know better and also have a clone, you can't compare the source code of the old firmware with the new one, what the frequency of the firmware is for the display and the new one with the old one Switching over and then creating a firmware file is just an idea. I would like to print faster but don't want to buy an original board and display for 150 euros. Maybe someone has an idea. Or an adapter between the cable and display fixes it with transistors,... there must be a solution, or Fysetc should find a solution.

KR4DIO, 2023-10-20:
> what the frequency of the firmware is for the display and the new one with the old one

I spent a little time looking through the source, but couldn't find where the SPI frequency was changed. However, as far as I can tell, the flashing doesn't actually affect printing. If I'm printing with the enclosure open and the flashing is annoying, I just go to Live Z Adjust mode, where the screen doesn't flash. It's not ideal, but at least for now the inconvenience is worth it for fast smooth prints.

pchrist, 2023-10-21:
I finally got hold of a Fysetc display, I hooked up a logic analyzer to the SPI signals between the MCU and the display controller. I found out that a read command to the display is issued every 2 seconds (on the SDA line). On a Prusa display the response is seen on the SDO line. On the  Fysetc display the response is seen on the SDA line. 

It looks like the Prusa display is running in full duplex mode and the Fysetc display is running in half duplex mode. As long as nothing is read back from the display this does not matter. Since release 4.6.1 a LCD communication check is executed every 2 seconds. The check will fail when running half duplex mode, resulting in a display reset, seen as a white screen. 

The half/full duplex configuration is done on the display by pulling some configuration pins high or low. These pins are not easily accessible on the Fysetc display.

diminDDL, 2023-10-21:
@pchrist could you share what the pins are even if they aren't easily accessible? Unless they are not broken out to the flat flex altogether.

CuriousTimo, 2023-10-21:
I'm looking at the st7789 datasheet and I think it had to do with IM3 pin.

SDA: 
-**When IM3: Low, SPI interface input/output pin**
-**When IM3: High, SPI interface input pin**

CuriousTimo, 2023-10-21:
Anyone able to compile from source code?

I'm getting an error: /Prusa-Firmware-Buddy-5.1.0-alpha1/utils/bootstrap.py", line 203, in switch_to_venv_if_nedded
    os.execv(str(venv_bin_dir / 'python'),
FileNotFoundError: [Errno 2] No such file or directory

I want to test out commenting out `reinit_lcd_and_redraw();` in src/gui/display_hw_checks_MINI.cpp
`void check_lcd() {
    uint8_t data_buff[ST7789V_MAX_COMMAND_READ_LENGHT] = { 0x00 };
    display::ReadMADCTL(data_buff);

    if ((data_buff[1] != 0xE0 && data_buff[1] != 0xF0 && data_buff[1] != 0xF8)) {
        reinit_lcd_and_redraw();
    }
}
} // anonymous namespace`

PhilippeGalid, 2023-10-22:
Thanks for your Info and Help,I hope Pruse see this and fixe this in the next Update.

Gabri988, 2023-10-22:
> I compiled and flashed version 5.1.0-alpha2
> 
> In file: src/gui/display_hw_checks_MINI.cpp I commented out line 21.
> 
> ```
> void check_lcd() {
>     uint8_t data_buff[ST7789V_MAX_COMMAND_READ_LENGHT] = { 0x00 };
>     display::ReadMADCTL(data_buff);
> 
>     if ((data_buff[1] != 0xE0 && data_buff[1] != 0xF0 && data_buff[1] != 0xF8)) {
>         //reinit_lcd_and_redraw();
>     }
> }
> } // anonymous namespace
> ```
> 
> My Fysetc 3.2" LCD is not flashing anymore.
> 
> The disadvantage is that you have to break out the tab on the Buddy board to run self-compiled firmware for this to work.

would you be so kind as to post your modified fw? I have the same fystec LCD and would like to solve the flashing problem

CuriousTimo, 2023-10-22:
To force a firmware update on the Mini:

- Push the reset-button, immediately followed by double-pressing the selector-knob.
- On the new screen, confirm the update by selecting the "FLASH" button and pressing the knob. 
[Mini 5.1.0-alpha2-no_reinit_lcd.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13062874/Mini.5.1.0-alpha2-no_reinit_lcd.zip)

Gabri988, 2023-10-22:
> [Mini 5.1.0-alpha2-no_reinit_lcd.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13062874/Mini.5.1.0-alpha2-no_reinit_lcd.zip)
> 
> To force a firmware update on the Mini:
> 
>     * Push the reset-button, immediately followed by double-pressing the selector-knob.
> 
>     * On the new screen, confirm the update by selecting the "FLASH" button and pressing the knob.

A thousand thanks! Just installed on buddy clone board without needing to cut the jumper

PhilippeGalid, 2023-10-22:
> > [Mini 5.1.0-alpha2-no_reinit_lcd.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13062874/Mini.5.1.0-alpha2-no_reinit_lcd.zip)
> > To force a firmware update on the Mini:
> > ```
> > * Push the reset-button, immediately followed by double-pressing the selector-knob.
> > 
> > * On the new screen, confirm the update by selecting the "FLASH" button and pressing the knob.
> > ```
> 
> A thousand thanks! Just installed on buddy clone board without needing to cut the jumper. it seems to work

Thanks for your Work and Help

PhilippeGalid, 2023-10-22:
> [Mini 5.1.0-alpha2-no_reinit_lcd.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13062874/Mini.5.1.0-alpha2-no_reinit_lcd.zip)
>
> To force a firmware update on the Mini:
>
>     * Push the [...]

Thanke for your Work and Help

diminDDL, 2023-10-22:
Can confirm that this fixes the flashes completely.

hreality, 2023-10-22:
Cofirmed.. flashing resolved with posted firmware

ZeekWorld, 2023-10-22:
Amazing work!

PhilippeGalid, 2023-10-22:
I've now tried flashing my Mini with the 2 files from the zip file, I press flash to start and then red screen Warning #12606 Firmware signature verification failed! What can i do? I have Fysetec Mini Clone with a Fysetc Board v1.0 and Screen ?

Lino77, 2023-10-22:
press ignore

Lino77, 2023-10-22:
@CuriousTimo The IM3 pin which is the one on the Fysetec display,
I would like to use the 3.2 display, but I don't want to break the seal, it´s on original Mini.

PhilippeGalid, 2023-10-22:
> press ignore

When i press ignore it start new with no Flashing

xXNoceboXx, 2023-10-23:
Thank you for that @CuriousTimo, it's works for me on Fystec 2.8 :)

nolookjones, 2023-10-23:
Thanks @CuriousTimo this fixed my Fystec flickering screen. Hopefully this makes it in the official version soon!

CuriousTimo, 2023-10-24:
> Hopefully this makes it in the official version soon!

I'm not optimistic about this. Prusa clearly added the display hardware checks for a reason to the latest firmware. It's a new feature, not a bug. 
And it is based on their LCD working in full duplex SPI mode. Where their display uses the SDO line to talk back to the Buddy board. The clone LCD screens seem to want to talk back on the SDA line.

I'm almost positive that an original Prusa LCD has IM3 pulled high and the clone LCDs do not. Then this behavior of the SPI SDO line is as expected and described in the ST7789V data sheet.

IM3 is not available on the flat cable and it should not need to be.
I'm hoping that it is available on the brown cable between the PCB and LCD panel.

But both the flat cable and the brown cable are 14 pins so it's hard to tell what connector [the documentation](https://github.com/prusa3d/Original-Prusa-MINI/blob/master/DOCUMENTATION/ELECTRONICS/mini-lcd-module-fw024tft-v14.pdf) is refering:
1 - LEDK
2 - LEDK
3 - LEDK
4 - LEDK
5 - LED-A
6 - GND
7 - RESET
8 - CS
9 - SCL
10 - RS
11 - SDA
12 - GND
13 - VCC
14 - GND

On some photos of the Fystec PCB, I can see R13 is not occupied, and am hoping it connects to pin 110 on the ST7789V chip.
But the BTT PCB looks different.

Anyway, I don't think lifting the LCD screen from the PCB to access the ST7789V chip is a viable option.

CuriousTimo, 2023-10-24:
Nevermind: I found the answer by looking at the gerber files:

Ribbon cable between LCD-PCB and Buddy Board:
1 - VDD 5V
2 - LCD_CS
3 - BTN_ENC
4 - LCD_RESET
5 - NETC7_1
6 - LCD_RS
7 - NETC9_1
8 - LCD_SCK
9 - KILL_PIN
10 - LCD_MOSI
11 - GND
12 - MISO
13 - GND
14 - BUZZER

So it seems that IM3 is not available on the brown cable from the PCB to the LCD screen :(

pchrist, 2023-10-25:
The 3.2" Fystec display uses a LCD module with a 20 wire cable. I have looked at various LCD modules with a ST7789V controller. I found this one on ALiExpress where the layout matches the one used by Fysetc:
https://vi.aliexpress.com/item/1005002266538639.html?gatewayAdapt=glo2vnm 

Here is the pinout of the cable: 
[image]
I have verified that GND,VDD,SDA,DC,CS and RST all are in the same location on the LCD module used by Fysetc. 
IM3 is not accessible, so full duplex mode can not be selected. This does not really matter since there is no SDO(MISO) signal available, so this module is half duplex only. 
I can not see any way to make the Fysetc display compatible by hardware modifications.

espinbolsa, 2023-11-03:
It works perfect for me. It doesn't blink anymore. But it doesn't let you change the **language**.
Thank you.

Alexender2k, 2023-11-06:
@CuriousTimo, thank you for the compiled bbf. Worked perfectly. I have a clone board, 3.2 Fytsetc LCD and all I did is to put the mini_release_boot.bbf on the flash drive, put it into the printer, pressed reset + quickly 3-4 times the knob, hit flash, hit ignore and voila, the screen is not flashing. There are some artefacts on the main screen icons (settings mainly) that appear (small yellow lines) but are redrawed after few moves on the main screen.
A big UP that nobody talks here is the fact the Input Shaper values CAN now be changed. After the flashing of your mini_release_boot.bbf, a new green menu option appeared ( Set Up values ) and you can tune the Input Shaper values if you wish to. Very nice!!!

CuriousTimo, 2023-11-18:
[5.1.0](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0-beta) Beta was released.

Compiled firmware for clone LCD users with flashing LCD issues:
[Mini 5.1.0-beta-no_reinit_lcd.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13399607/Mini.5.1.0-beta-no_reinit_lcd.zip)

```
To force a firmware update on the Mini:
* Push the reset-button, immediately followed by double-pressing the selector-knob.
* On the new screen, confirm the update by selecting the "FLASH" button and pressing the knob.
```

Fuzarca, 2023-11-22:
@CuriousTimo Thank you very much!!! This is awesome. It appears to work perfect in my clone.

CuriousTimo, 2023-11-24:
[5.1.0](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0) Final was released.

Compiled firmware for clone LCD users with flashing LCD issues:
[Mini 5.1.0-release-no_reinit_lcd.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13457212/Mini.5.1.0-release-no_reinit_lcd.zip)

```
To force a firmware update on the Mini:
* Push the reset-button, immediately followed by double-pressing the selector-knob.
* On the new screen, confirm the update by selecting the "FLASH" button and pressing the knob.
```

espinbolsa, 2023-11-24:
Thank you very much for all your efforts

Trek3D6, 2023-11-26:
>Compiled firmware for clone LCD users with flashing LCD issues: [Mini 5.1.0-release-no_reinit_lcd.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13457212/Mini.5.1.0-release-no_reinit_lcd.zip)

Will this fix an early first run original mini that has the same flashing issue?

caracolmaker, 2023-11-28:
> [5.1.0](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0) Final was released.
>
> Compiled firmware for clone LCD users with flashing LCD issues: [Mini [...]

Thank you very much for your help.
If you could make a video of how to do it, it would be great for the future.

[121 more replies]

closed 2025-03-21 by Prusa-Support