prusa3d/Prusa-Firmware-Buddy #3278: [BUG] Divide by 0 hardfault when disconnecting USB micro-B at printer

murk-sy, 2023-09-19:
**Printer type** - MINI

**Printer firmware version** - 5.1.0-alpha

**Original or Custom firmware** - Original/Custom

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint**
Pronterface via microusb cable

**Describe the bug**
After ungracefully removing the USB cable at the printer end, the printer appears to hardfault at the next function ran on the printer itself.

**How to reproduce**
1. Reset printer
2. Connect to a PC
3. Connect to printer using Pronterface
4. Run G28 and wait for it to finish
5. Disconnect the micro-B USB cable *at the printer* - disconnecting at the PC (USB-A) does not reproduce the bug, nor does gracefully closing the serial connection
6. Run Control->Auto Home
7. Printer will immediately reset and bluescreen

[image]

Reproduced on 2 different minis.

**Expected behavior**
No issue when continuing to use the printer (as when physically disconnecting a USB-A cable).

Likely almost impossible to run into this issue during normal use, but I happened to while testing things.

**Crash dump file**
[dump_buddy.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12664336/dump_buddy.zip)

xtomasnemec, 2023-09-21:
This happens on every version of the firmware

Prusa-Support, 2023-09-30:
Thanks for reporting and I can confirm this issue was already there before FW 5.1.0-alpha1.

Please allow me to close this issue as a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2972.

Michele Moramarco
Prusa Research

closed 2023-09-30 by Prusa-Support