prusa3d/Prusa-Firmware-Buddy #3282: [BUG] <Enter comprehensive title>

OlNissen, 2023-09-20:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI+]

**Printer firmware version** - [5.1.0]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**USB drive

**Describe the bug**
   A clear and concise description of what the bug is.

**Expected behavior**
  print

**G-code**
   Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
   Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
   Please attach a video. It usually helps to solve the problem.
[D.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12671177/D.zip)

misan, 2023-09-20:
Same problem here. Core dump and gcode file are attached.

[Archivo comprimido.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12675435/Archivo.comprimido.zip)

jwidess, 2023-09-20:
This is pretty much the same issue as [#3275](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3275) and should probably be merged.

OlNissen, 2023-09-20:
solved

I could not print Shaper from USB

But try to format the USB and make it from scratch and then it ran again.

misan, 2023-09-20:
Just deleting files from the USB made it possible to print the same file. I did not need to reformat the pendrive. It had more than one hundred g-code files before and that was causing the problem.

closed 2024-05-30 by danopernis