prusa3d/Prusa-Firmware-Buddy #3283: [BUG] 5V Voltage on Prusa XL shows 33V on XL

schwabe, 2023-09-20:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - XL

**Printer firmware version** - 4.7.2

**Original or Custom firmware** - Original

**Optional upgrades** - Stock

**Describe the bug**
  A value for 5V in the info sensor screen on XL shows an unlikely high value.

**How to reproduce**
  navigate to the sensor info screen, scroll down to 5V. This has been confirmed on two other XL.

**Expected behavior**
  The displayed voltage should much loser to 5V.

**Picture**
[image]

BrienKing, 2023-09-24:
I saw that as well.

Voxel3DPrinting, 2023-10-23:
I have the same issue

closed 2024-06-25 by github-actions[bot]