prusa3d/Prusa-Firmware-Buddy #3285: [BUG]  5.0.0 New smaller font decreases readability

the-powl, 2023-09-20:
**Printer type** - MK4

**Printer firmware version** - 5.0.0

**Original or Custom firmware** - [Original/Custom]

**Describe the bug**
  The font that was introduced with FW 5.0.0. is smaller than the ones in previous releases. In my opinion it decreases the readability by quite an amount because the font is too thin for the resolution of the display. The anti aliasing of the font leads to too much blurriness.

**How to reproduce**
  Upgrade to 5.0.0. Take a look in the main menu at for example the letter h in "Preheat". The vertical lines of the h are not even fully white, they consist of pure blur.

**Expected behavior**
  Bigger, non-blury font. (The font in previous releases was just fine imho)

RainerGa, 2023-11-24:
The information bar should have a larger font. Alternatively, there could be a configuration in which you can choose between two font sizes.

RainerGa, 2024-06-01:
Did anybody know if prusa is working on this issue? Thanks for replies.

danopernis, 2024-06-01:
No, at this point this isn't being worked on.

Disclaimer, following is my personal opinion:

I do not want this issue to be automatically closed, because I think it is important. Some usability improvements would be nice especially for visually impaired folks. It could also set us apart from the competition.

Technically, this should be feasible, but it would require substantial rework of the internal layout subsystem which is already clunky due to sharing the codebase between printers with ili9488 and st7789v display drivers.

andreacampanella, 2024-06-04:
I agree, this is very important.

closed 2026-10-05 by github-actions[bot]