prusa3d/Prusa-Firmware-Buddy #3286: [BUG] <Seam transition catastrophic>

mozgy, 2023-09-20:
**Printer type** - [MK4]

**Printer firmware version** - [5.0.0]

**Original or Custom firmware** - [Original]

**Optional upgrades** - []

**USB drive or USB/Octoprint** - USB flash drive

**Describe the bug**
  Seam transition is catastrophic, way too late start of extrusion so huge gaps -
[image]
[image]

**How to reproduce**
  Probably any combination of multiple objects with only one perimeter (like vase mode).

**Expected behavior**
  Perfect print

**G-code**
**Crash dump file**
**Video**

murk-sy, 2023-09-20:
This is likely not a firmware issue, but more related to gcode generation and material tuning.

You should provide a sample project so it's easier to check if it's actually reproducible on other machines.

mozgy, 2023-09-20:
@murk-sy You might be right, resliced it with nonIS profile and quite a bit better but still not perfect

[image]
[image]

[IssueWithMK4.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12678492/IssueWithMK4.zip)
Should I reopen this issue in PrusaSlicer section ?

darraghbr, 2023-09-21:
At least some of these parts appear to be a single perimeter thick, if that is the case you should be using spiral vase mode instead of a conventional print.

murk-sy, 2023-09-21:
As a side note since you didn't mention the manufacturer, some white filaments can be more problematic to print than others. If other filaments print fine, it's just the white filament that may require some extra tuning.

mozgy, 2023-09-22:
> At least some of these parts appear to be a single perimeter thick, if that is the case you should be using spiral vase mode instead of a conventional print.

In case you did not notice, there are *several* models *together* on the plate as *that* combination shows the problem. Spiral vase mode works fine for single model.

darraghbr, 2023-09-22:
> > At least some of these parts appear to be a single perimeter thick, if that is the case you should be using spiral vase mode instead of a conventional print.
> 

> In case you did not notice, there are _several_ models _together_ on the plate as _that_ combination shows the problem. Spiral vase mode works fine for single model. Since it works fine for nonIS sliced it is **huge** regression.

I did notice that there is more than one part on the build plate. What you might not realise is that you can use sequential printing to print multiple objects in vase mode on a single build plate.

[image]

mozgy, 2023-09-22:
> I did notice that there is more than one part on the build plate. What you might not realise is that you can use sequential printing to print multiple objects in vase mode on a single build plate. [...]

You're right, I could've, but 'Issues' section is about bugs and regressions not walkarounds hence specific setup ..

FutureSharks, 2023-12-19:
I opened this issue for PrusaSlicer: https://github.com/prusa3d/PrusaSlicer/issues/11914

Maybe it's the same issue? Not sure if it's a Slicer or firmware problem because they were both released and updated together in order to get the input shaper feature.

GithubUser99999999, 2024-05-09:
still not fixed by Prusa

danopernis, 2024-07-09:
This is not a firmware bug. As already suggested, this could relate either to slicer or the material settings.

closed 2024-07-09 by danopernis