prusa3d/Prusa-Firmware-Buddy #3287: [BUG] Unable to login to PrusaLink on Safari (macOS/iOS/iPadOS )

pyrho, 2023-09-20:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [ALL]

**Printer firmware version** - [>= 5.0.0]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [N/A]

**USB drive or USB/Octoprint** - USB drive

**Describe the bug**
The `nonce` generated by the backend for HTTP Digest authentication (used by PrusaLink) is only valid for 5 seconds (as defined in the code [here](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/bfb0ffc745ee6546e7efdba618d0e7c0f4c909cd/src/common/http/types.h#L151)).

Safari (contrary to the other browsers it seems) does not re-submit the credentials with a fresh nonce upon receiving a 401 response code. So each time Safari displays its Digest Auth modal window, you **only have 5 seconds** to provide valid credentials, after which the `nonce` will become stale and any login attempt will fail.

This seems to be specific to Safari, as it would seem like the other browsers work around this by re-submitting the authentication request when the `nonce` is stale (using the freshly provided one).

Short of submitting a feature request to Safari, our only hope (assuming we're sticking with Digest Auth),  is to increase the `nonce`'s lifetime to something more sensible. [jvasileff](https://github.com/jvasileff) mentioned in https://github.com/prusa3d/Prusa-Link-Web/issues/386 that [Apache's default `nonce` lifetime is 300](https://httpd.apache.org/docs/2.4/mod/mod_auth_digest.html#authdigestnoncelifetime). This document also states the following:

> this should probably never be set to less than 10 seconds

This timeout is too short, especially on iOS where typing is slower than on a desktop, making it impossible to login in the 5 second time frame allowed.

**How to reproduce**
Try to connect to PrusaLink (FW >= 5.0.0) using Safari (iOS, macOS, or iPadOS), take more than 5 seconds to enter your login and password when prompted; the login attempt will fail.

**Expected behavior**
Users are able to log into PrusaLink using the Safari web browser.

**How to fix**

Patch provided by  [jvasileff](https://github.com/jvasileff) in https://github.com/prusa3d/Prusa-Link-Web/issues/386#issuecomment-1727946657

```diff
diff --git i/src/common/http/types.h w/src/common/http/types.h
index 2c8a42b4..e5ce21b8 100644
--- i/src/common/http/types.h
+++ w/src/common/http/types.h
@@ -148,7 +148,7 @@ using Url = std::array<char, MAX_URL_LEN>;
 // hopefully never gets stale nonce for request uploading a gcode, which
 // can cause an infinit upload loop, if the browser does not read errors
 // before sending the whole body.
-static const uint32_t valid_nonce_period = 5;
-static const uint32_t extended_valid_nonce_period = 8;
+static const uint32_t valid_nonce_period = 300;
+static const uint32_t extended_valid_nonce_period = 300;
 
 } // namespace http
```

**G-code**
N/A

**Crash dump file**
N/A

**Video**

Typing the credentials after 5 seconds demo (fail):
[image]

Typing the credentials within 5 seconds demo (success):
[image]

amoose136, 2023-09-22:
Good find. Hopefully a fix is soon in the main release.

Claghorn, 2023-09-28:
Just finished my MK4 kit build, loaded firmware 5.0.0, setup wifi, but I can't login with any browser. I just keep getting the login window over any over again.

pyrho, 2023-09-29:
> Just finished my MK4 kit build, loaded firmware 5.0.0, setup wifi, but I can't login with any browser. I just keep getting the login window over any over again.

I've seen many reports of people unable to use any Safari-based browser, but they were all able to login with Chrome or Firefox (on desktop that is, not on iOS).
Are you sure you are not mistyping your password ? 😅

Prusa-Support, 2023-10-18:
Thank you very much for the suggestion.

I'll make sure the developers will acknowledge this thread.

I wonder if this can be escalated as a Pull Request or if it can be closed as a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3043 and https://github.com/prusa3d/Prusa-Link-Web/issues/386.
In the first case though, the issue should be closed anyway to open a new Pull Request instead.

Michele Moramarco
Prusa Research

pyrho, 2023-10-18:
Hi @Prusa-Support 

Thank you for your reply, #3043 is definitely the same issue as this one (sorry I should have commented on it instead of opening this issue).

I've gone ahead and opened a PR: https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/3371/files with the fix.

Please don't hesitate to close this,  https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3043 and https://github.com/prusa3d/Prusa-Link-Web/issues/386 if you think it's best.

samirkouider, 2023-11-20:
Just managed to replicate this on safari for mobile. Waited more than 5s and as usual credentials prompt just reappeared. Copied my password for next attempt, for less than 5s, pasted and logged in no problem

GRIFFCOMMca, 2023-11-20:
Just upgraded to v5.1.0 (Mk4), seem to have the same issue in Windows
- Windows 10
- Edge, Chrome, Firefox

srp3rd, 2023-11-25:
I've run into the same problem.  I can't consistently log in using Safari, which I was able to do before I upgraded the firmware.  Upvote for this fix.

Prusa-Support, 2023-11-25:
@GRIFFCOMMca I'm not sure if this issue affects Windows and Chrome as they are the most popular combination but I don't seem to see similar reports. Also, this issue is quite specific about Safari (macOS/iOS/iPadOS).
Please make sure your computer is connected to the same local network as your printer (Prusa Link is available only on the local network) and double check IP/key via Prusa Connect (the latter allows remote connection outside the local network).

@pyrho Thanks for your commitment and for opening the Pull Request.

This issue will be closed because opened after issue #3043.
Please monitor PR https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/3371 or feel free to comment on the initial issue mentioned above.

Michele Moramarco
Prusa Research

closed 2023-11-25 by Prusa-Support

emc02, 2024-09-11:
opened a new issue, because it's not related to Safari and only happens on opening the API-Sites, but I think it's related to this bug: [#507](https://github.com/prusa3d/Prusa-Link-Web/issues/507)