prusa3d/Prusa-Firmware-Buddy #3288: [BUG] 5.1.0-alpha issue and error

muitobem, 2023-09-21:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - 5.1.0-allpha

**Original or Custom firmware** -i  optional upgrade
**Optional upgrades** - [e.g. Filament Runout Sensor]

**USB drive or USB/Octoprint
  Please let us know if you print via USB flash drive 

**Describe the bug**
  no preview image on the MINI LCD of the selected sliced model to print.  Also error and crash half way through the print

**How to reproduce**  Just try to print the model again
  
**Expected behavior**
  A clear and concise description of what you expected to happen.  I expected and LCD preview and completion of the print

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error  

[image]

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug

[image]

**Video**
  Please attach a video. It usually helps to solve the problem.

murk-sy, 2023-09-21:
Please provide the 3mf project file and gcode file, and also which version of Prusaslicer were you using - makes it a lot easier for devs to reproduce the issues.

muitobem, 2023-09-21:
Thank you for your reply.
I selected the incorrect MINI configuration in Prusa Slicer.  The MINI Firmware was correct, but the correct Prusa Slicer configuration was not selected.

closed 2024-06-25 by github-actions[bot]