prusa3d/Prusa-Firmware-Buddy #3289: V5.0.0 Does mnot detect XY end contacts anymore.

rocrail, 2023-09-21:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MK4]

**Printer firmware version** - [5.0.0]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [none]

**USB drive or USB/Octoprint**
  Prusa Link

**Describe the bug**
  After flashing V5.0.0 the print startup does not detect the XY endcontacts and almost destroyed my MK4 !!!

**How to reproduce**
  Just install V5.0.0 and print.

marek26340, 2023-09-25:
Have you tried doing a full selftest? A LOT has changed in the 5.0.0 firmware, including the way that the MK4 homes it's axis.
Also, **check your belt tension** using the official web app here: https://belt.connect.prusa3d.com/ **before** doing anything else, including the selftest I mentioned above. The Prusa printers have apparently gotten waaay more sensitive to having just a bit too loose/tight belts.

Starting a print immediately after a firmware upgrade on a Prusa is not the greatest idea in the first place. Especially since they started using the Buddy boards in them. Not implying they're bad, it's just that they always change lots of stuff under the hood on them with each release...

rocrail, 2023-09-25:
It would be an idea to write in the readme.txt that the Selftest is always needed after a firmware upgrade.
If the current printing is ready I will try this and report back.

rocrail, 2023-09-25:
OK, that did help. No the Homing is as expected.
Please add this remark in the readme file and on the Firmware Download page at Prusa:
https://help.prusa3d.com/de/downloads

Prusa-Support, 2023-12-28:
Thanks for your input.
As highlighted in the comments above and in https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3215#issuecomment-1871427830, so far the problem seems to be a matter of belt tensioning, or other hardware-related problems.

Please allow me to close this issue as a partial duplicate and potentially solved at hardware level.

Michele Moramarco
Prusa Research

closed 2023-12-28 by Prusa-Support