prusa3d/Prusa-Firmware-Buddy #3290: [QUESTION] Prusa MINI 5.1.0-alpha1 maximum nozzle temperature question

Gyrxiur, 2023-09-21:
Not sure if bug or feature, but with new firmware, max nozzle temp is 275 °C, with older it is 290 °C. In nozzle replacement guide it says to preheat to 280 °C, and that is not possible with 5.1.0-alpha1 FW. So I guess it is a bug, or not? If not, what is reason for it?

marek26340, 2023-09-25:
Same issue. Just tried setting 290°C via Gcode and the temp got set back down to 275°C. 
@Prusa-Support Could you please give us an explanation for this? Thank you!

allengregoryiv, 2023-11-27:
Has this ever been resolved it looks like the official release also caps at 275c from what I have seen.

allengregoryiv, 2023-11-28:
[image]

Attempting to load PA which defaults to 285C the printer refuses to heat past 275C.

Noxmiles, 2023-12-27:
Just installed Firmware 5.1.2 from 4.4.1. Manual Nozzle max temperature is 275°. Even the guide for replacing the heatblock says it must be set to 280° - which is not possible anymore!

https://help.prusa3d.com/de/guide/auswechseln-des-heizblock-heatbreak-mini-mini_122769

closed 2024-06-25 by github-actions[bot]