prusa3d/Prusa-Firmware-Buddy #3291: [FEATURE REQUEST] Option to set "Printer ready" on printer

rdaek, 2023-09-21:
**Printer type** - [MK4]

Menu item (maybe inside Network->Connect) to be able to set printer as  "Ready" for printing next gcode in queue without need to do it via browser.

murk-sy, 2023-09-21:
In my mind, something like this in the Print menu also makes sense (since it's a fairly standard procedure)

> ↩ Return
> 🇨 Connect: Set ready!
> File1.gcode
> File2.gcode
> ...

Ideally, the same button should be available on the print finished screen.
However, since Connect integration will (as far as I know) start collecting data if print is successful or not, finishing a print will likely be a 2-3 step process.

Skaronator, 2023-12-27:
This is now available with the 5.1.0 release:

> Added the option to set the printer’s status as "ready" for Prusa Connect directly from the printer menu

https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0

dpark12300, 2024-01-04:
Someone close the issue. It's on there already

Photograaf16, 2024-01-12:
> In my mind, something like this in the Print menu also makes sense (since it's a fairly standard procedure)
>
>
> Ideally, the same button should be available on the print finished screen. However, [...]

100% Agree, it should be on the "Finished" screen where you can select "retry" and "home" i believe... there is always 1 grayed out, replaced that one with "set ready".

dpark12300, 2024-01-12:
I second this

closed 2024-06-25 by github-actions[bot]