prusa3d/Prusa-Firmware-Buddy #3292: [BUG] "Imprecise data access" BSOD when printing - MINI 5.1.0-alpha

matt-chapman, 2023-09-21:
**Printer type** - MINI+

**Printer firmware version** - 5.1.0-alpha

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint** - USB

**Describe the bug**
  A few minutes into the print, the printer started emitting a repeating beep, and the screen displayed an error message (image attached)
[image]

**How to reproduce**
  Unknown - subsequent print attempt succeeded.

**Expected behavior**
  The print completes, without producing an error.

**G-code**
  [calibration-cube_0.4n_0.2mm_PLA_MINIIS_19m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12682699/calibration-cube_0.4n_0.2mm_PLA_MINIIS_19m.gcode.zip)

**Crash dump file**
  [dump_buddy.bin.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12682709/dump_buddy.bin.zip)

**Video**
  No video available

Thanks for your efforts with this new release!

marek26340, 2023-09-26:
Off topic - May I ask why are you printing a cali cube? Do you think that your printer is printing dimensionally inaccurate parts? Tightening the belts to have a spot-on tension and squaring the printer should be plenty enough in my experience...

matt-chapman, 2023-09-26:
> Perhaps this was actually an error regarding the flash drive?

Certainly could be - though I'll add I am using a new flash drive and not the one included with the printer. I'm hopeful that the crash dump will provide answers to someone equipped to make sense of it :)

> Off topic - May I ask why are you printing a cali cube? Do you think that your printer is printing dimensionally inaccurate parts? Tightening the belts to have a spot-on tension and squaring the printer should be plenty enough in my experience...

No particular reason, I'm just experimenting with the new firmware and printing little widgets just for the fun of watching it go!

closed 2024-05-30 by matt-chapman