prusa3d/Prusa-Firmware-Buddy #3294: [BUG] [5.1.0-alpha] Out of heap memory

Traumlewis, 2023-09-21:
**Printer type** - [MINI+]

**Printer firmware version** - [5.1.0-alpha]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [none]

**USB drive or USB/Octoprint**
  USB drive: Kingston DataTraveler 100 G3 32GB (used with Prusa Connect)

**Describe the bug**
  Out of heap memory blue screen 
   (heap:cpp.37)

**How to reproduce**
1. take PrusaSlicer 2.6.1 alpha profil 
2. Slice [Bambu Lab AMS Rear Dessicant Tray Lid v3.STL and Bambu Lab AMS Rear Dessicant Tray v3.STL](https://www.printables.com/de/model/409676-bambu-lab-ams-rear-desiccant-refillable-tray/remixes)
3. Export G-code 
4. uploade to Prusa Connect.
5. and Print (not)

**Expected behavior**
  not printing 

**G-code**
[Bambu Lab AMS Rear Dessicant Tray v3_0.4n_0.2mm_PETG_MINIIS_59m.gcode](https://gofile.io/d/LHDzgi)

**Crash dump file**
[dump_buddy.bin](https://gofile.io/d/LHDzgi)

**Photo**
[20230921_164023](https://gofile.io/d/LHDzgi)

jasonehines, 2023-10-17:
I've also gotten an Out of heap memory crash. I sliced (Version 2.6.1+MacOS-arm64) a model I created and exported it to a usb drive (Lexar E32C 128GB Type-C USB Flash Drive). The printer was preheating for PETG when I inserted the usb drive and started the print, I then got the BSOD. I'm attaching the gcode and crash dump.

[Mini Crash.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12934995/Mini.Crash.zip)

KAJIKK, 2023-11-05:
Also gotten this after i uploaded bgcode file from prusaslicer.
[image]
[dump_buddy.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13260911/dump_buddy.zip)

mix579, 2023-11-17:
Also getting apparently random Out of Heap Memory errors on my Minis, printing from USB stick.
[
[dump_buddy.bin.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13392399/dump_buddy.bin.zip)
](url)

velkoborsky, 2024-03-31:
Similar for me in v 5.1.2+13478. BSOD while downloading from Prusa Connect. Caused a failed print as the download started while printing the previous object. The blue screen said task:connectTask rather than media_prefetch as above.

danopernis, 2024-06-26:
Let me close this as a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2351

MINI is low on RAM, we are aware of that and taking steps to improve memory allocation patterns, it just takes long :disappointed:

closed 2024-06-26 by danopernis