prusa3d/Prusa-Firmware-Buddy #3297: [BUG] MK4 with 5.0.0 encountered Watchdog Reset task:defaultTask?

craigktreasure, 2023-09-21:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK4

**Printer firmware version** - 5.0.0

**Original or Custom firmware** - Original

**Optional upgrades** - None

**USB drive or USB/Octoprint**

USB drive

**Describe the bug**

I left a print running during the night. I have printed this particular model\gcode several times. I believe the print completed successfully, but the next morning the printer was beeping with a blue screen indicating an error.

[image]

I reset the printer and it asked if I wanted to create a dump, which I did (see attached).

**How to reproduce**

Not sure.

**Expected behavior**

No blue screens should be encountered.

**G-code**

In case it's useful, this is the last g-code I printed prior to encountering the error.
[catapultcard_square165_prusaslicer_0.4n_0.2mm_PLA_MK4IS_2h13m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12693280/catapultcard_square165_prusaslicer_0.4n_0.2mm_PLA_MK4IS_2h13m.gcode.zip)

**Crash dump file**

[dump_buddy.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12693264/dump_buddy.zip)

**Video**

None

Velocity11, 2023-10-08:
Having the same issue at the moment. Are there any fixes yet?

sauloquirino, 2023-10-13:
I'm having the same error in my Mini+

Mr2201, 2023-10-19:
I'm also encountering the same issue with my MK4. The error suddenly appeared after I had successfully printed my specific object once. I attempted it twice, and the error occurred around 20 minutes into the print (2 hours 20 minutes total print time) both times. After that, I restarted the printer (power cycle) and tried again. This time, the crash happened approximately 5 minutes before the print finished. It seems there's no specific time when the issue occurs. For all prints, including the successful one, I used RepetierServer as host. Tomorrow, I'll try it again from the original USB stick to see if that changes anything.

jurriaan, 2023-10-31:
Just had a similar issue (WATCHDOG RESET task:defaultTask?) with 5.1.0-alpha2 on my MK4. The printer finished printing 5-6 hours before the issue appeared (was just turned on without doing anything before crashing).

7s3, 2023-11-12:
This happened the me 3-5 times over the course of the past week.
MK4 version 5.0.0 and 5.0.1

F1dla, 2023-12-12:
Same here on self upgraded MK3.9.
[image]

craigktreasure, 2024-05-30:
Not stale

danopernis, 2024-05-30:
@craigktreasure did you test with the latest firmware? Can you provide crash dump?

craigktreasure, 2024-05-30:
> @craigktreasure did you test with the latest firmware? Can you provide crash dump?

I have not reproduced the issue with the latest firmware. There's a dump in my original post.

danopernis, 2024-05-31:
Yes, I know that there is a dump in the original post, but that crash happened on the v5.0.1 There were several bugfixes related to watchdog and its interaction with the buggy USB sticks since that version, so that's why I am asking you to provide crash dump (if any) from the latest firmware. Otherwise, I do not understand your earlier "Not stale" comment.

craigktreasure, 2024-05-31:
Well, from my perspective, there was no indication that anyone had acknowledged or seen the issue. A bot came through saying, "i'm going to close this making it even less likely that they'll see it." So, I said it wasn't stale hoping that it still might get some attention.

Now that's you're acknowledging the issue and indicating that there might have been related fixes, we can certainly close it and re-activate it if we see the issue again in newer firmware.

closed 2024-05-31 by craigktreasure