prusa3d/Prusa-Firmware-Buddy #3300: [BUG] Blue Screen when gcode file uploaded over PrusaLink - Dashboard

matejus, 2023-09-22:
blue screen when uploaded file over PrusaLink - dashboard page. File successfully uploaded to the USB flash drive, then printer failed to BSD (expect to prepare file to print). 
Printer is not connected to internet (time is not set from timeserver)
g-code prepared by PrusaSlicer 2.6.1

**Printer type** - MINI
**Printer firmware version** - 5.1.0.alpha1+12820
**USB drive or USB/Octoprint** -USB Flash drive

[gcode_dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12698289/gcode_dump.zip)

darraghbr, 2023-09-22:
Was the bluescreen image the same as the one in this issue?

#3299

matejus, 2023-09-22:
Hi, not sure, but i think the screen was full of text. but i don't have photo

strugatzki, 2023-09-22:
I have the same issue as OP, it was definitely not the bluescreen referred to by darraghbr  above (ie, not a question of previews), but a uhm fully fledged BSOD with lots of text. OK I'll take a pic next time. Occurs when sending a file directly from PrusaSlicer to MINI+, both unter Input Shaper Alpha and with a non Input Shaper-Printer Profile. When saving the same gcode to a USB drive, it works, including a correct preview pic.

closed 2024-06-25 by github-actions[bot]